MarlinFirmware/Marlin #26742: [BUG] (No serial connection via SERIAL_PORT_3 on STM32F1-board BTT SKR mini E3 V2.0)

marioqxx, 2024-01-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have self-compiled Marlin-FW 2.1.2.1 with default configuration. I have only #define SERIAL_PORT_3 1 (and PORT_2) configured compared to default. According the documentation, for STM32F1-boards serial connection through PORT_3 shall work. This includes the BTT SKR mini E3 v2.0 board (also compliant to the PR here: https://github.com/MarlinFirmware/Marlin/pull/21784).

I have tried to connect with Octoprint, but in terminal it says no response, time-out. I have then attached an oscilloscope. The serial command is sent (signal is fine and shall be recognized as serial signal by the board), but no response is visible. I have also tried Serial-Print M-code, no visible response on the oscilloscope either. The board is otherwise connected through TFT-serial connection and works fine with the LCD at 250000 baud-rate and also through USB-serial.

In [Marlin/src/pins/stm32f1/pins_BTT_SKR_MINI_E3_V2_0.h](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/pins/stm32f1/pins_BTT_SKR_MINI_E3_V2_0.h), lines 51 and 52, the pinout for the EXP1-connector is shown. All PIN-assignments appear to be correct according the schematics, i.e. pin 3 connected with PA9 and pin 5 connected with PA10 - which are the RX1 and TX1 pins of the STM32F1, but both pins are labeled with "BTN_EN1" and "BTN_EN2", instead of RX1 and TX1 as in the BTT SKR-mini manual (https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0%20Instruction%20Manual.pdf) page 7.

Use-case is as in the above referenced PR, attaching an raspberry PI Zero 2W running Octoprint and providing fast WIFI.

### Bug Timeline

This is a new bug, I havn'et found any related bug-reports.

### Expected behavior

Configure with #define_SERIAL_PORT_3 1 the serial interface and serial connection via EXP1 connector to Octoprint shall work.

### Actual behavior

As descibed above. No response visible on oscilloscope.

### Steps to Reproduce

Configure SERIAL_PORT_3 1, compile Marlin 2.1.2.1 and flash board. Connect Raspi through serial connection and connector EXP1 of the SKR Mini E3 V2.0 board.

### Version of Marlin Firmware

2.1.2.1

### Printer model

Creality Ender-5 with BTT SKR Mini E3 V2.0 Board

### Electronics

BTT SKR Mini E3 V2.0 Board

### LCD/Controller

BTT TFT35 E3 V3.0 Touch Screen

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

None. Configuration used is Default provided with Marlin and only change is SERIAL_PORT_3 1.

ellensp, 2024-01-28:
If your talking about this https://github.com/MarlinFirmware/Configurations/blob/release-2.1.2.1/config/examples/Creality/Ender-5/BigTreeTech%20SKR%20Mini%20E3%202.0/Configuration.h

you need to disable #define CR10_STOCKDISPLAY
As that redefines the RX1/TX1 IO pins as BTN_EN1 and BTN_EN2

marioqxx, 2024-01-29:
@eensp: Thanks for the quick reply and the answer that I thought should solve the isse. I did disable CR10_STOCKDISPLAY and some accopanying settings SHOW_CUSTOM_BOOTSCREEN, CUSTOM_STATUS_SCREEN_IMAGE so that compile runs through. Flashed the new firmware, but still haven't received any response. So unfortunately not working yet.

I received an EEPROM CRC-check fail, then re-flashed and no such error. I do not expect this error has anything to do with the serial-issue here, I would think this has only something to do with EEPROM where parameters are stored and if the error persisted I would have Reset EEPROM with M5x-command. I'll have a closer look later and re-check my wiring, but there are only a very limited number of possibilities for error, only RX and TX-wires.

@config: Yes, right, this is the config I refer to.

ellensp, 2024-01-29:
the board a a number of build environment. of the following which one did you use?

STM32F103RC_btt
STM32F103RC_btt_USB
STM32F103RE_btt
STM32F103RE_btt_USB 
STM32F103RC_btt_maple 
STM32F103RC_btt_USB_maple 
STM32F103RE_btt_maple
STM32F103RE_btt_USB_maple

marioqxx, 2024-01-29:
I did use: STM32F103RC_btt_USB. I have the "RC"-variant of the STM32-chip.

marioqxx, 2024-02-01:
I thought to update my testing, in case anyone else has a similar problem, he can read here some way to hunt down issues.

According my research, the M118 command should be able to test the TX-line of the board. I did so through the attached BTT-screen and its built-in terminal. I did not receive any response though.

There is also a separate way to test the RX-line. This can be accomplished with connecting the Raspi with installled octoprint for example. The open up a ssh-connection to octoprint and insert (you may replace the baud-rate with the one you set in your firmware Configuration.h):

sudo stty 115200 < /dev/ttyS0
echo -ne 'G28\n' > /dev/ttyS0

In my case this started Auto Home. Alternatively the same can be accomplised with minicom.

sudo apt install minicom
sudo minicom -b 115200 -D /dev/ttyS0

In the minicom-console enter G28 and press return. To exit minicom use CTRL-A   Z   X.

In my case, this indicates that the TX1-line may have more likely than not a Hardware-issue. I have ordered another board, so I can flash the same firmware and see if that works. In that case, any firmware-issue is ruled out.

marioqxx, 2024-02-10:
UPDATE: I've changed the board for a new BTT SKR mini E3 V2.0, flashed 2.1.2.2 (STM32F103RC_btt_USB) with //#define CR10_STOCKDISPLAY disabled as suggested by @ellensp. Octoprint did connect to /dev/ttyS0 without any issue. I have not tested with Marlin 2.1.2.1, because if did not compile with the latest PlatformIO on my system, but would not expect any different outcome.

Please close this issue, because this was related to an HW-issue of the TX-line of my board.

I would however suggest adding the information provided from @ellensp to the config-file, so that for this board, when SERIAL_PORT_3 is enabled, it is stated that this requires disabling CR10_STOCKDISPLAY. For consistency it could make sense to have the pin's for EXP1-connector in the board's pin-definition having this also documents along with the RX1/TX1-pins shown as in the HW-manual.

closed 2024-02-10 by marioqxx