MarlinFirmware/Marlin #26743: [BUG] (bug summary)

nare809, 2024-01-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

there is no Ender 3 printer with Creality 4.2.2 Board compiler environment of ARM - STM32F103RET6. please add it

### Version of Marlin Firmware

2.1.2.1

### Printer model

Ender 3

### Electronics

Creality 4.2.2 with STM32F103RET6 Environment

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Crazy-Charles, 2024-01-28:
You haven't included  a ZIP file containing your Configuration.h and Configuration_adv.h but that configuration is available

#define MOTHERBOARD BOARD_CREALITY_V422

thisiskeithb, 2024-01-28:
You'd use one of these environments:

- STM32F103RE_creality
- STM32F103RE_creality_xfer
- STM32F103RC_creality
- STM32F103RC_creality_xfer
- STM32F103RE_creality_maple

https://github.com/MarlinFirmware/Marlin/blob/bf8675b44a43c82a1ee8595fa3f7a1f1e5dca61e/Marlin/src/pins/pins.h#L655-L656

closed 2024-01-28 by thisiskeithb