MarlinFirmware/Marlin #26760: [FR] support accelerations higher than 65 535 (16 bits)

phouille, 2024-02-01:
### Is your feature request related to a problem? Please describe.

disclaimer : I am in mechanics and therefore have a very limited knowlege about marlin, computer code & many other related subjects

I am working on a proof of concept concerning very fast printing on a marlin printer (no klipper no raspberry just the mainboard), but I struggle to set an acceleration above 65535mm/s2 as the value gets back to 0 after that. 
I understand this limit as the acceleration value being stored in a 16 bit variable but I very much could be wrong. 
- I have not defined "LIMITED_MAX_ACCEL_EDITING"
- I have both tried setting the accel value below 65535 then cranking it up via the LCD menu and setting the accel value above 65535 in the firmware with the same results
- I have not done any measure of said acceleration, I just looked at the value from the LCD screen (meaning it could just be the LCD part of the code being unable to display variables above 2^16)

for some reason I also can't set the junction deviation value above 0.300 but I can live without for the moment

### Are you looking for hardware support?

I'm running marlin 2.1.2.1 with input shaping and linear advance enabled using the config files from [Ender3/BigTreeTech SKR Mini E3 3.0](https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%203.0) on an SKR mini E3 V3 and my plateform is a slighltly modified ender3

### Describe the feature you want

help on how to set accelerations above 2^16mm/s2 and/or support in marlin for accelerations above this value, if possible up to 19 bits as my final goal is to get to 300 000mm/s2 with extremely light axis

I also believe this feature to be useful to the general public, as said above I am running my firsts test on an ender3 from 2018, the X axis is worse from stock (same 42-34 stepper but heavier print head) and this is the axis I managed to push over 65k accel with well calibrated input shaping and driver currents.

### Additional context

feel free to ask me any marlin file that could help you, here's my current "configuration.h" and "configuration_adv.h" (but in .txt), every line that I edited has a " //ED " comment
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/14119431/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/14119433/Configuration_adv.txt)

ellensp, 2024-02-01:
Please learn basic gcode

eg https://marlinfw.org/docs/gcode/M201.html

If you set M201 X128000  it takes it fine
you can then check it with M201 

It is not stored in a 16bit value. it is a unsigned 32bit value

This is a LCD issue

All these User interfaces are for normal daty to day users, not for anyone attempting anything out of the ordinary, For that you  
should take direct control of the machine via gcode.

ellensp, 2024-02-01:
you need to make these changes

```DIFF
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 9a0cd1addc..3999198c97 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -1261,9 +1261,9 @@
  */
 #define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 100, 10000 }
 
-//#define LIMITED_MAX_ACCEL_EDITING     // Limit edit via M201 or LCD to DEFAULT_MAX_ACCELERATION * 2
+#define LIMITED_MAX_ACCEL_EDITING     // Limit edit via M201 or LCD to DEFAULT_MAX_ACCELERATION * 2
 #if ENABLED(LIMITED_MAX_ACCEL_EDITING)
-  #define MAX_ACCEL_EDIT_VALUES       { 6000, 6000, 200, 20000 } // ...or, set your own edit limits
+  #define MAX_ACCEL_EDIT_VALUES       {900000, 900000, 900000, 20000 } // ...or, set your own edit limits
 #endif
 
 /**
diff --git a/Marlin/src/lcd/menu/menu_advanced.cpp b/Marlin/src/lcd/menu/menu_advanced.cpp
index 840b658d09..3f258db679 100644
--- a/Marlin/src/lcd/menu/menu_advanced.cpp
+++ b/Marlin/src/lcd/menu/menu_advanced.cpp
@@ -523,7 +523,7 @@ void menu_backlash();
     // M204 T Travel Acceleration
     EDIT_ITEM_FAST(float5_25, MSG_A_TRAVEL, &planner.settings.travel_acceleration, 25, max_accel);
 
-    #define EDIT_AMAX(Q,L) EDIT_ITEM_FAST_N(long5_25, _AXIS(Q), MSG_AMAX_N, &planner.settings.max_acceleration_mm_per_s2[_AXIS(Q)], L, max_accel_edit_scaled[_AXIS(Q)], []{ planner.refresh_acceleration_rates(); })
+    #define EDIT_AMAX(Q,L) EDIT_ITEM_FAST_N(long8_25, _AXIS(Q), MSG_AMAX_N, &planner.settings.max_acceleration_mm_per_s2[_AXIS(Q)], L, max_accel_edit_scaled[_AXIS(Q)], []{ planner.refresh_acceleration_rates(); })
     NUM_AXIS_CODE(
       EDIT_AMAX(A, 100), EDIT_AMAX(B, 100), EDIT_AMAX(C, 10),
       EDIT_AMAX(I,  10), EDIT_AMAX(J,  10), EDIT_AMAX(K, 10),
diff --git a/Marlin/src/lcd/menu/menu_item.h b/Marlin/src/lcd/menu/menu_item.h
index 823d2a4a25..e76aebda75 100644
--- a/Marlin/src/lcd/menu/menu_item.h
+++ b/Marlin/src/lcd/menu/menu_item.h
@@ -170,6 +170,7 @@ DEFINE_MENU_EDIT_ITEM_TYPE(float51sign ,float    ,ftostr51sign    ,  10        ,
 DEFINE_MENU_EDIT_ITEM_TYPE(float52sign ,float    ,ftostr52sign    , 100        , + 0.001f ); // +123.45
 DEFINE_MENU_EDIT_ITEM_TYPE(long5       ,uint32_t ,ftostr5rj       ,   0.01f               ); // 12345      right-justified
 DEFINE_MENU_EDIT_ITEM_TYPE(long5_25    ,uint32_t ,ftostr5rj       ,   0.04f               ); // 12345      right-justified (25 increment)
+DEFINE_MENU_EDIT_ITEM_TYPE(long8_25    ,uint32_t ,ftostr8rj       ,   0.04f               ); // 12345678   right-justified (25 increment)
 
 #if HAS_BED_PROBE
   #if WITHIN(PROBE_OFFSET_ZMIN, -9, 9)
diff --git a/Marlin/src/libs/numtostr.cpp b/Marlin/src/libs/numtostr.cpp
index 6f4e4a480b..fe50ddf0b5 100644
--- a/Marlin/src/libs/numtostr.cpp
+++ b/Marlin/src/libs/numtostr.cpp
@@ -44,6 +44,7 @@ constexpr long UINTFLOAT(const float V, const int N) {
 }
 
 char conv[9] = { 0 };
+char conv2[17] = { 0 };
 
 // Format uint8_t (0-100) as rj string with __3% / _23% / 123% format
 const char* pcttostrpctrj(const uint8_t i) {
@@ -117,6 +118,29 @@ inline const char* ui16tostrXrj(const uint16_t xx, const int index) {
   return &conv[index];
 }
 
+// Convert unsigned 32bit int to right-justified string
+inline const char* ui32tostrXrj(const uint32_t xx, const int index) {
+  switch (index) {
+    case 0 ... 3: conv2[3]  = RJDIGIT(xx, 1000000000000);
+    case 4:       conv2[4]  = RJDIGIT(xx, 100000000000);
+    case 5:       conv2[5]  = RJDIGIT(xx, 10000000000);
+    case 6:       conv2[6]  = RJDIGIT(xx, 1000000000);
+    case 7:       conv2[7]  = RJDIGIT(xx, 100000000);
+    case 8:       conv2[8]  = RJDIGIT(xx, 10000000);
+    case 9:       conv2[9]  = RJDIGIT(xx, 1000000);
+    case 10:      conv2[10] = RJDIGIT(xx, 100000);
+    case 11:      conv2[11] = RJDIGIT(xx, 10000);
+    case 12:      conv2[12] = RJDIGIT(xx, 1000);
+    case 13:      conv2[13] = RJDIGIT(xx, 100);
+    case 14:      conv2[14] = RJDIGIT(xx, 10);
+  }
+  conv2[15] = DIGIMOD(xx, 1);
+  return &conv2[index];
+}
+
+// Convert unsigned 16bit int to string with 12345 format
+const char* ui32tostr8rj(const uint32_t xx) { return ui32tostrXrj(xx, 16 - 8); }
+
 // Convert unsigned 16bit int to string with 12345 format
 const char* ui16tostr5rj(const uint16_t xx) { return ui16tostrXrj(xx, 8 - 5); }
 
@@ -353,6 +377,12 @@ const char* ftostr5rj(const_float_t f) {
   return ui16tostr5rj(i);
 }
 
+// Convert unsigned float to rj string with 12345 format
+const char* ftostr8rj(const_float_t f) {
+  const long i = UINTFLOAT(f, 0);
+  return ui32tostr8rj(i);
+}
+
 // Convert signed float to string with +123.45 format
 const char* ftostr52sign(const_float_t f) {
   long i = INTFLOAT(f, 2);
diff --git a/Marlin/src/libs/numtostr.h b/Marlin/src/libs/numtostr.h
index fde07e8368..02f4bf324f 100644
--- a/Marlin/src/libs/numtostr.h
+++ b/Marlin/src/libs/numtostr.h
@@ -107,6 +107,9 @@ const char* ftostr54sign(const_float_t x, char plus=' ');
 // Convert unsigned float to rj string with 12345 format
 const char* ftostr5rj(const_float_t x);
 
+// Convert unsigned float to rj string with 12345678 format
+const char* ftostr8rj(const_float_t x);
+
 // Convert signed float to fixed-length string with +12.3 / -12.3 format
 const char* ftostr31sign(const_float_t x);
 
```

ellensp, 2024-02-01:
you do need to enable LIMITED_MAX_ACCEL_EDITING or marlin uses a internal default of 99000

phouille, 2024-02-01:
thanks a lot, I'll be trying these solutions and provide updates on whether or not I did get it to work. 

Forgot to write it in the first message but I did try the different commands related to accelerations including M201, I just wasn't aware of the LCD menu limits (suspected it tho) 

one question, you did a great job making the needed modifications stand out to me but I'm confused about the meaning of these kinds of lines : 

diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 9a0cd1addc..3999198c97 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -1261,9 +1261,9 @@

do I understand correctly "--- a/Marlin/Configuration.h  +++ b/Marlin/Configuration.h" as making a B version of Configuration.h ? I'm probably wrong about this but there's no "a" or "b" folder before the "Marlin" one

thanks again for your patience, as said above I am a mechanics so anything code related still is a bit cryptic to me but I am learning (still we're the ones providing the community with motors, belts & new designs and I think it's great that everyone has their own speciality)

phouille, 2024-02-01:
thanks !! so do the lines : 

diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 9a0cd1addc..3999198c97 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h

just mean I don't edit the file name ? if yes then I'll just edit each file manually in vscode and build marlin

thinkyhead, 2024-02-02:
I'm not sure why a DIFF was posted as a comment, as we really should just make that into a PR. As part of that patch, we can also automatically detect that the max value is large and make the number wider based on that. I've got a small pile of tasks tonight, but if no one else creates the PR then I can go ahead and take care of that soon.

ellensp, 2024-02-02:
This is such a obscure use case I have zero intention of adding it to the code base for all users

ellensp, 2024-02-02:
I do however think the menu code should not roll over like this, it should get to max editable value and stop

phouille, 2024-02-02:
thanks a lot guys it works perfectly and I can edit values straight from the LCD !! (the modifications to Marlin/src/lcd/menu/ should have given me a hint) 

my second smaller problem still remains tho as I can't edit the junction deviation value over 0.3, I've tried M205 J0.31 with the following results : 

>>> M205 J0.31
SENDING:M205 J0.31
Error:?J out of range (0.01 to 0.3)

is this range defined by the mathematics behind JD (from what I've read I don't see why on the physics side but computers are again not my world) or has 0.3 just been chosen as a higher limit to prevent hardware degradation ? if so would you have any idea for a solution to modify this limit ? 

anyway thanks a lot for your time and work, as said above I do think that the max LCD values for acceleration can nowaday become a bit limiting since the implementation of input shaping (which works absolutely great, well done) and I would love to see these changes in a public release but you are the juges here

closed 2024-02-29 by thinkyhead