MarlinFirmware/Marlin #26766: [BUG] (Z OFFSET not responding)

kbritt02, 2024-02-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm running a CR touch on an Ender 3. I have an upgraded board 4.2.7. I wanted to add more mesh points and upgraded Marlin to Marlin-bugfix-2.1.x I'm now having an issue with the Z probe being set not working. It is not adjusting at all. I tried saving and reloading as well as tuning while printing, and nothing seems to fix this. Do you have any suggestions on a fix?

### Version of Marlin Firmware

Marlin-bugfix-2.1.x

### Printer model

Ender-3 

### Electronics

board 4.2.7

### Other add-ons

CR Touch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/14151781/Configuration.zip)

steljwagh, 2024-02-05:
I have the exact same 4.2.7 board and I just installed the lastest fw. Can you give more details about the steps you follow in the menu? Because my z-offset settings are working fine from the prepare / z-offset menu.

I have however an issue after automatic leveling, the printer gets stuck in the menu without saving the new mesh. I'll investigate later and maybe open a bug for that.

steljwagh, 2024-02-05:
I have no issues at all with the latest firmware, just flashed the latest firmware.

Setup z-offset, with use of all possible resources in the menu.
Did a bed-level (although I had to do it from octoprint, because bedleveling doesn't save from printer menu anymroe)
Started print and tried to alter z-offset from tuning menu (working)

Are you sure your gcode is not messing with your printer settings?

closed 2024-02-05 by ellensp