MarlinFirmware/Marlin #26767: MKS Tinybee motherboard, mks h43 touch screen, error when compiling

Alphilippe, 2024-02-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Unable to compile,i don't know why,work with 2.0 and not with 2.1

### Version of Marlin Firmware

2.1

### Electronics

tinybee

### LCD/Controller

mks h43

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/14152410/Configuration.zip)

ellensp, 2024-02-04:
error is 

```
Marlin/src/lcd/extui/dgus/DGUSDisplay.cpp: In static member function 'static void DGUSDisplay::initDisplay()':
Marlin/src/lcd/extui/dgus/DGUSDisplay.cpp:70:3: error: 'LCD_SERIAL' was not declared in this scope
   LCD_SERIAL.begin(LCD_BAUDRATE);
   ^

```

thisiskeithb, 2024-02-04:
`LCD_SERIAL_PORT` is also disabled, but it should be set to `2` for this board.

It won't fix the error, but the TinyBee should be added here:

https://github.com/MarlinFirmware/Marlin/blob/7f4792e47c909cef2b46a11a0a8a80cf6858b80d/Marlin/src/inc/Conditionals_adv.h#L1279-L1280

ellensp, 2024-02-04:
please try this, it compiles

In Configuration.h 
under 
`  #define MOTHERBOARD BOARD_MKS_TINYBEE
`add
```
  #define LCD_SERIAL_PORT 2
  #define HARDWARE_SERIAL2_RX 16
  #define HARDWARE_SERIAL2_TX 17

```
and in Marlin/src/HAL/ESP32/HAL.h

under
```
#elif ENABLED(WIFISUPPORT)
  #define MYSERIAL2 webSocketSerial
#endif
```
add 

```
#ifdef LCD_SERIAL_PORT
  #if !WITHIN(LCD_SERIAL_PORT, 2, 2)
    #error "LCD_SERIAL_PORT must be 2."
  #endif
  #define LCD_SERIAL Serial2
  #if HAS_DGUS_LCD
    #define LCD_SERIAL_TX_BUFFER_FREE() LCD_SERIAL.available()
  #endif
#endif
```

If this works, ill look at formalizing it and creating a PR

Alphilippe, 2024-02-04:
Yes,it was the error,now it compile fine,i have uploaded the firmware to the printer and also work .
Thank you.

closed 2024-02-04 by Alphilippe