MarlinFirmware/Marlin #26768: [FR] "Replace Nozzle" LCD Menu Command

rbuckton, 2024-02-03:
### Describe the feature you want

Every so often I need to replace the nozzle on my hot end, as one does, and there are often a lot of steps involved to do this correctly (removing the sock, heating the nozzle, removing the filament, raising the extruder, etc.). If replacing the nozzle is something you do somewhat infrequently, it's fairly easy to forget a step. This often means having to wait for the nozzle to cool down or heat up again if, for example, you forget to remove the sock.

To make this easier for myself, I wrote a gcode file that automates the various steps I would follow to replace the nozzle on my Ender 3 Pro w/sprite extruder. While this is a fine solution for me, I'm wondering if it would be beneficial for Marlin to have this as a built-in feature for others.

The nozzle change script I use can be found here: https://gist.github.com/rbuckton/93060c4c628235ac5382ca477a072fa3
Some of the gcode commands are specific to my printer setup (filament unload/load distance, purge amount, etc.) and use material settings for PLA (via `M109 I0`) so you end up needing multiple scripts if you work with multiple materials.

thisiskeithb, 2024-02-03:
You can do this with `CUSTOM_MENU_MAIN`:

https://github.com/MarlinFirmware/Marlin/blob/9e21330d7ae8b4979fee10099294e9e61c939613/Marlin/Configuration_adv.h#L4005-L4013

closed 2024-02-03 by thisiskeithb

rbuckton, 2024-02-03:
Would that not still only be specific to my own custom builds? Does that mean there's no interest in a general purpose solution for a wider audience?

EvilGremlin, 2024-02-03:
Looks like solution in search of a problem, and 7 prompts is plain obstructive. But to each their own, this is what custom menu items exist for.