MarlinFirmware/Marlin #26772: [BUG] (Encoder stopped working after complying firmware)

Eviotie, 2024-02-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I got an BL touch so I decided to get some new firmware with it. I complied my own firmware for the GD32F303 RET6 (linked bellow). Next I installed it on my ender 3 4.22v but then i get an error eeprom loop, after some research I was told to use the commands M502, and M500. On first glance that appeared to work but as soon as I tried going into the UI, nothing would happen. To clarify If i twist the encoder nothing would happen or if I click the enonder nothing would happen.

https://drive.google.com/drive/folders/1bYn1BO3WZwIFL3fKtjB9zwTOD3IiYG1j?usp=sharing ( Link of firmware)

M115 output

FIRMWARE_NAME:Marlin bugfix-2.1.x (Feb  4 2024 19:55:10) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 4.2.2 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_POS:0
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:RUNOUT:1
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:HOST_ACTION_COMMANDS:0
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:1
Cap:MULTI_VOLUME:0
Cap:REPEAT:0
Cap:SD_WRITE:1
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:0
Cap:LFN_WRITE:0
Cap:CUSTOM_FIRMWARE_UPLOAD:0
Cap:EXTENDED_M20:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:1
Cap:EP_BABYSTEP:0
Cap:CHAMBER_TEMPERATURE:0
Cap:COOLER_TEMPERATURE:0
Cap:MEATPACK:0
Cap:CONFIG_EXPORT:0

### Version of Marlin Firmware

Marlin bugfix-2.1.x

### Printer model

Creaitly ender-3

### Electronics

Stock but GD32F303 RET6

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

>>> M111 S247
SENDING:M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
>>> M105
SENDING:M105
echo:M105
ok T:18.59 /0.00 B:18.76 /0.00 @:0 B@:0

[config_h_and_adv.zip](https://github.com/MarlinFirmware/Marlin/files/14160918/config_h_and_adv.zip)

https://drive.google.com/drive/u/0/folders/1bYn1BO3WZwIFL3fKtjB9zwTOD3IiYG1j whole file

ellensp, 2024-02-06:
There are 5 build environment options to choose from, which of the following  did you select?

STM32F103RE_creality
STM32F103RE_creality_xfer
STM32F103RC_creality
STM32F103RC_creality_xfer
STM32F103RE_creality_maple

ellensp, 2024-02-06:
Verified there are no changes in your archive from marlin b12340b1d57288945633ad926136ab2c27d25c12 
Other than the expected Configs file 

Tested your archive on build environment STM32F103RE_creality on a creality 4.2.2 with a GD32F303 RET6 MPU and a CR10_STOCKDISPLAY

The encoder works as expected.

ellensp, 2024-02-06:
Make sure you reset eeprom back to defaults. Ideally by enabling  EEPROM_INIT_NOW

closed 2024-02-06 by ellensp

Eviotie, 2024-02-07:
> Make sure you reset eeprom back to defaults. Ideally by enabling  EEPROM_INIT_NOW
> Or by gcode M502 (Factory Reset) then M500 (Save Settings)

I tryed both of theses things, the first one restarted the printer on boot the second one said something along the lines of Epprom saved, and also I'm using the normal 512kb enviorment

Eviotie, 2024-02-07:
Nvm I recompiled it again and it worked thank you very much for you help