MarlinFirmware/Marlin #26776: [BUG] Compilation error: #error "GRID_MAX_POINTS_[XY] must be a whole number between 3 and 15."

WillShawx, 2024-02-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Doesn't compile due to error in ABL Grid sizing

### Bug Timeline

new - today

### Expected behavior

I expected the FW to compile

### Actual behavior

error occurred - Compilation error: #error "GRID_MAX_POINTS_[XY] must be a whole number between 3 and 15."

### Steps to Reproduce

attempt to compile attached config 

### Version of Marlin Firmware

2.1.2.1

### Printer model

Custom build

### Electronics

MKS Gen V1.4

### LCD/Controller

RepRap Discount Full Graphic Smart Controller

### Other add-ons

BLTouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/14194695/Marlin.zip)

plampix, 2024-02-07:
On line 1979 of Configuration.h:
#elif ENABLED(AUTO_BED_LEVELING_)

Should be
#elif ENABLED(AUTO_BED_LEVELING_UBL)

ellensp, 2024-02-07:
Yes this is a user generated issue

closed 2024-02-07 by ellensp