MarlinFirmware/Marlin #26778: [BUG] TS35 + Robin Nano V3.1 Garbling display

jonhysfl, 2024-02-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When the screen is touched, the interface gets a little funky, with random colours.
There's a video here, that shows the exact behaviour.

https://github.com/MarlinFirmware/Marlin/assets/91577386/1092a1fe-74c0-4fbc-8078-6074cb5fd99f

[Sapphire_SP5-V3.zip](https://github.com/MarlinFirmware/Marlin/files/14202740/Sapphire_SP5-V3.zip)

### Expected behavior

Expected the screen to work just as my elegoo neptune, with marlin 2.1.2.1, without errors.

### Actual behavior

Random colours appearing ate any touch.

### Version of Marlin Firmware

2.1.2.1

### Printer model

Sapphire SP5-V3

### Electronics

Robin Nano V3.1

### LCD/Controller

TS35-R

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-02-07:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

jonhysfl, 2024-02-08:
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

Oh man, thank you.
Sorry for the "unecessary" bug report.

This solved almost all of the issues, sometimes, i have the same bug, but changing to the nightly build solved 95% of that.

I was scared because the endstops changed a bit in this version, but in the end, it was very easy to dial everything.

I created a repository with the configuration for this version of the printer, for anyone that wants to dial their changes, can start from a working point:
https://github.com/jonhysfl/SP5-V3-Marlin

Thanks!

closed 2024-02-08 by thisiskeithb