MarlinFirmware/Marlin #26781: [BUG] Auto bed leveling error

Pezsma, 2024-02-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello, 
I wanna use BLTouch with my 3d printer with AUTO_BED_LEVELING_BILINEAR.
Leveling completes, but while printing it makes errors, like it wasn't properly leveled, sometimes with bigger differences.
Z offset is good, and NOZZLE_TO_PROBE_OFFSET is set precisely.

There is no mechanical error, nothing is lose or I have no similar errors in my setup.
I also warm everything before G29 to printing temperature.
I ran the Z_MIN_PROBE_REPEATABILITY_TEST, and seems fine.

READ: Mean: 0.009500 Min: -0.005 Max: 0.035 Range: 0.040
READ: Standard Deviation: 0.010764

The machine is a costum built one with mega 2560 and ramps 1.4. Drivers are TMC2208 with serial connected too.
I have lead screws on all axes.

I attached my configuration and I tried withthe latest source from lts-2.1.1 and bugfix-2.1.x too. Also attached some screenshots.

If you got any ideas what i messed up, please help me out.

### Bug Timeline

uset this as the first fw for this printer

### Expected behavior

After auto bed leveling is should print based on that data.

### Actual behavior

Doesn't follow the bed based on the auto bed leveling result. It changes the levels, but sometimes it seems like it is mirrored, sometimes simply makes +-1mm diff.

### Steps to Reproduce

Auto bed level, then start print (from the same gcode file)
[config.zip](https://github.com/MarlinFirmware/Marlin/files/14220822/config.zip)
[image]

### Version of Marlin Firmware

lts-2.1.1

### Printer model

Custom

### Electronics

ramps1.4

### LCD/Controller

-

### Other add-ons

TMC2208

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-02-09:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2024-02-09 by ellensp

htotoo, 2024-02-14:
If you have the #define MIN_SOFTWARE_ENDSTOPS it'll work like this.
Even if you have:
#define Z_PROBE_LOW_POINT          -10  
#define Z_PROBE_OFFSET_RANGE_MIN -20
#define Z_PROBE_OFFSET_RANGE_MAX 20

Just disable this:
//#define MIN_SOFTWARE_ENDSTOPS
and everything should work with your bed leveling correctly.

There could be another solution, but this works at least.