MarlinFirmware/Marlin #26784: [BUG] UBL MESH_MIN_Y does not work, uses X values

JPT77, 2024-02-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am currently setting up auto leveling using the Creality BLtouch on two old Creality printers. 
It was hard work and I just added a pull request for the UBL docs. 

There is still a a problem that looks like a bug. 
I want to have the grid leave room at both ends of the Y axis +-15mm so that it doesn't collide with the clamps.
And I want to have the grid at 5 and 30 mm at both ends of the X axis, because the probe offset is 45 mm. 
X works fine, even moving the hotend 15 mm beyond the bed.

For the X axis this works well but for Y it just uses the X values. or something like 5/30.

`
#if EITHER(MESH_BED_LEVELING, AUTO_BED_LEVELING_UBL)
  // Override the mesh area if the automatic (max) area is too large
  // JPT this is where G29 should probe. If a point is beyond probe margins it cannot be reached, and the value will be empty (.)
  #define MESH_MIN_X 5
  #define MESH_MIN_Y 15
  #define MESH_MAX_X X_BED_SIZE - 30
  #define MESH_MAX_Y Y_BED_SIZE - 15
#endif
`

ps. this seems to affect G26 as well.
[Ender3Pro-config.zip](https://github.com/MarlinFirmware/Marlin/files/14232798/Ender3Pro-config.zip)
[ender.log](https://github.com/MarlinFirmware/Marlin/files/14232800/ender.log)

### Bug Timeline

I don't know. did not use previous marlin versions

### Expected behavior

expected to leave 10 to 15 room in Y axis direction while probing UBL to not collide with clamps.

### Actual behavior

left 5 / 30 mm room

### Steps to Reproduce

G28 
G29 P1

### Version of Marlin Firmware

2122 and 2123 bugfix git

### Printer model

Creality Ender-3 Pro

### Electronics

Creality STM32 427

### LCD/Controller

stock Creality display

### Other add-ons

BLtouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I had some issues with serial comms and reduced to 57600 baud, which works fine.

classicrocker883, 2024-02-15:
would something like this work? https://github.com/MarlinFirmware/Marlin/pull/26761

basically in the variable mesh inset feature, you're allowed to edit the `MESH_INSET` for each side.  
which effectively changes the probed edge so you can avoid bed clamps.  
you're able to adjust one side, or all 4, from the max bed size to as little of a mesh as you'd like.

go ahead and copy the **PR** source code and test it out if you want.

JPT77, 2024-05-17:
I think I solved it. 
forgot how ;)
so just closing.

closed 2024-05-17 by JPT77