MarlinFirmware/Marlin #26786: [WEBSUPPORT ON ESP3D_WIFISUPPORT] (WEBSUPPORT working on 2.1.2.1 but no in 2.1.2.2)

Skeleton84, 2024-02-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Some change in Configuration_adv.h, in new version now no chance to enable WEBSUPPORT, OTASUPPORT on tinybee (ESP3D_WIFISUPPORT)
only change is #if EITHER(WIFISUPPORT, ESP3D_WIFISUPPORT) to now: #if ENABLED(WIFISUPPORT)
Code works fine but without webserver.
How to enable WEBSUPPORT, OTASUPPORT without errors?

In 2.1.2.2.
//#define WIFISUPPORT         // Marlin embedded WiFi management. Not needed for simple WiFi serial port.
#define ESP3D_WIFISUPPORT   // ESP3D Library WiFi management (https://github.com/luc-github/ESP3DLib)

/**
 * Extras for an ESP32-based motherboard with WIFISUPPORT
 * These options don't apply to add-on WiFi modules based on ESP32 WiFi101.
 */
#if ENABLED(WIFISUPPORT)
  #define WEBSUPPORT          // Start a webserver (which may include auto-discovery) using SPIFFS
  #define OTASUPPORT          // Support over-the-air firmware updates
  #define WIFI_CUSTOM_COMMAND // Accept feature config commands (e.g., WiFi ESP3D) from the host
  //AUTHENTICATION_FEATURE: protect pages by login password.
  #define AUTHENTICATION_FEATURE
  /**
   * To set a default WiFi SSID / Password, create a file called Configuration_Secure.h with
   * the following defines, customized for your network. This specific file is excluded via
   * .gitignore to prevent it from accidentally leaking to the public.
   *
   *   #define WIFI_SSID "WiFi SSID"
   *   #define WIFI_PWD  "WiFi Password"
   */
  //#include "Configuration_Secure.h" // External file with WiFi SSID / Password
#endif

in 2.1.2.1.

//#define WIFISUPPORT         // Marlin embedded WiFi management
#define ESP3D_WIFISUPPORT   // ESP3D Library WiFi management (https://github.com/luc-github/ESP3DLib)

#if EITHER(WIFISUPPORT, ESP3D_WIFISUPPORT)
  #define WEBSUPPORT          // Start a webserver (which may include auto-discovery)
  #define OTASUPPORT          // Support over-the-air firmware updates
  #define WIFI_CUSTOM_COMMAND // Accept feature config commands (e.g., WiFi ESP3D) from the host
  //AUTHENTICATION_FEATURE: protect pages by login password.
  #define AUTHENTICATION_FEATURE
  /**
   * To set a default WiFi SSID / Password, create a file called Configuration_Secure.h with
   * the following defines, customized for your network. This specific file is excluded via
   * .gitignore to prevent it from accidentally leaking to the public.
   *
   *   #define WIFI_SSID "WiFi SSID"
   *   #define WIFI_PWD  "WiFi Password"
   */
  //#include "Configuration_Secure.h" // External file with WiFi SSID / Password
#endif

### Version of Marlin Firmware

2.1.2.2.

### Printer model

Creality Ender-3 Pro

### Electronics

mks-tinybee

### LCD/Controller

cr10_stockdisplay

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/14236649/Marlin.zip)

thisiskeithb, 2024-02-12:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

Skeleton84, 2024-02-13:
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

yes i tried this today again with your link, but webserver not working, same as version 2.1.2.2, now i revert back to 2.1.2.1 and webserver working fine.

PanosPetrou, 2024-02-17:
I can confirm this too. I have a Tinybee board too, I had to change #if ENABLED(WIFISUPPORT) to #if ANY(WIFISUPPORT, ESP3D_WIFISUPPORT) to make it work. 
#if EITHER(WIFISUPPORT, ESP3D_WIFISUPPORT) will work too.

thisiskeithb, 2024-02-26:
I've opened https://github.com/MarlinFirmware/Marlin/pull/26822 with a fix.

closed 2024-02-26 by thisiskeithb

kyon03, 2024-03-22:
i tried this out still on latest bugfix still not working

> Same issue here - same board (WiFi set as AP). The fix above doesn't work on the latest bugfix release. I had to do this:
> 
> ```c
> /**
>  * Native ESP32 board with WiFi or add-on ESP32 WiFi-101 module
>  */
> //#define WIFISUPPORT         // Marlin embedded WiFi management. Not needed for simple WiFi serial port.
> #define ESP3D_WIFISUPPORT   // ESP3D Library WiFi management (https://github.com/luc-github/ESP3DLib)
> 
> /**
>  * Extras for an ESP32-based motherboard with WIFISUPPORT
>  * These options don't apply to add-on WiFi modules based on ESP32 WiFi101.
>  */
> #if ANY(WIFISUPPORT, ESP3D_WIFISUPPORT)
>   #define WEBSUPPORT          // Start a webserver (which may include auto-discovery) using SPIFFS
>   // #define OTASUPPORT          // Support over-the-air firmware updates
>   // #define WIFI_CUSTOM_COMMAND // Accept feature config commands (e.g., WiFi ESP3D) from the host
> 
>   /**
>    * To set a default WiFi SSID / Password, create a file called Configuration_Secure.h with
>    * the following defines, customized for your network. This specific file is excluded via
>    * .gitignore to prevent it from accidentally leaking to the public.
>    *
>    *   #define WIFI_SSID "WiFi SSID"
>    *   #define WIFI_PWD  "WiFi Password"
>    */
>   //#include "Configuration_Secure.h" // External file with WiFi SSID / Password
> #endif
> ```
> 
> Sanitycheck throws errors if `OTASUPPORT` or `WIFI_CUSTOM_COMMAND` are enabled.

thisiskeithb, 2024-04-28:
Patch for `2.1.x` submitted:
- https://github.com/MarlinFirmware/Marlin/pull/27025