MarlinFirmware/Marlin #26788: [FR] Input Shaping - support more then one frequency

vovodroid, 2024-02-12:
### Is your feature request related to a problem? Please describe.

Resonance could be on more then one frequency (e.g. belt and rails), but only one of them could be dumped.

### Describe the feature you want

Support resonance dumping on several frequencies.

tombrazier, 2024-05-05:
If there is more than one resonant frequency, ZV still gives some value by using the average of the two frequencies. For a wider range, general practice is to use a different input shaping algorithm. In principle these can be implemented the same way as ZV if someone has the time to do it. But if you're on 32 bit it might be worth trying FT_MOTION as that does have other input shaping algorithms.

tombrazier, 2024-05-07:
It's under `FT_MOTION` in `Configuration_adv.h`. And see https://marlinfw.org/docs/gcode/M493.html.

vovodroid, 2024-05-07:
Yes, I've seen it, still no details. Are there some reviews of real world experience?

tombrazier, 2024-05-07:
I haven't seen any 3D printing news websites or youtubers reporting on it (but I haven't really been looking either). It's a newer feature. I don't know how much uptake it has had yet. I think it's a case of give it a try and see. Or, if you're not keen on being bleeding edge, then waiting.

I think the theory behind the FT_MOTION feature is fundamentally sound, though, and for faster MCUs it may be the future.

In other news I know how to extend Marlin's "classic input shaping" to other algorithms and I am hoping I can persuade someone to do it.

vovodroid, 2024-05-08:
> EI or ZVD would be good algorithms to try out. If those are not good enough, ZVDD and 2HEI are next, etc.

Actually Ulendo opposites their "vibration compensation" to these algorithms

https://static1.squarespace.com/static/5d6c3f03a45b79000185fcd2/t/651eb5d6b97671504ecf42a0/1696511447068/Input+Shaping+vs+Ulendo+VC.pdf

https://static1.squarespace.com/static/5d6c3f03a45b79000185fcd2/t/6514bea50bb58c52a358f45b/1695858342352/Ulendo+Input+Shaping+Comparison+White+Paper+2023.pdf

Is M493  S1 Fixed-Time Motion mode this "vibration compensation" algorithm?

vovodroid, 2024-05-08:
So no one knows what are M493 parameters))). I'll try to ask in original PR.

tombrazier, 2024-05-08:
> So no one knows what are M493 parameters))). I'll try to ask in original PR.

I didn't say that. But do ask in the original PR and tag me and I will attempt some kind of reply. I know a bit about FT_MOTION.

vovodroid, 2024-05-08:
Also what is difference between default M900 Linear Advance and FT MOTION one?