MarlinFirmware/Marlin #26792: [BUG] Delta goes insane

rq3, 2024-02-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The latest (02/13/24) bugfix goes nuts on my delta. All steppers, including the extruder, move wildly and randomly. M48 and G33 are not possible.

### Bug Timeline

Very recently. I currently run bugfix from Jan 12, 2024 very well.

### Expected behavior

I expect it to do what the g-code tells it to do.

### Actual behavior

It does whatever it wants.

### Steps to Reproduce

1) Download, compile, and install the latest bugfix
2) Run known good g-code
3) Dive for power switch

### Version of Marlin Firmware

2.1.2.2

### Printer model

Anycubic Predator

### Electronics

stock Trigorilla Pro (STM32F103ZETA

### LCD/Controller

Anycubic TFT 3.5

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/14272957/configs.zip)

rq3, 2024-02-15:
> could you drop the max feed rate to something like 1mm/min and then demonstrate 'insane'? then also a slow crawl to the power switch should suffice

Nope, that's not it. Attached are the configs for my working copy of bugfix 2.1.2.1
Not much difference per WinMerge comparison, and the feedrate is even higher.
[2.1.2.1_Working_Configs.zip](https://github.com/MarlinFirmware/Marlin/files/14300741/2.1.2.1_Working_Configs.zip)

ellensp, 2024-02-15:
Lowering max feedrate  was so you could observe what it is doing...  so we have a clue where to start looking.
goes insane is not a useful description.

rq3, 2024-02-15:
Here;s a couple of short videos, running identical Benchy g-code:
[Movs.zip](https://github.com/MarlinFirmware/Marlin/files/14302286/Movs.zip)

rq3, 2024-05-20:
The 2.1.2.2 bugfix on May 20, 2024 works, kinda, so I am closing this. I would still like to see the missing significant figures for calibration standard deviations, but that's a related but different issue.

closed 2024-05-20 by rq3