MarlinFirmware/Marlin #26797: Motherboard won't communicate with tft neither Pronterface with enabled Input shaping

WenzelHyde, 2024-02-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

With enabled input shaping there is no communication with tft display or pronterface. I attached my modified config files, but it is the same with original example files...

### Bug Timeline

Started with 2.1.2.2, worked on 2.1.2.1

### Expected behavior

I expected that motherboard will communicate with display and Pronterface.

### Actual behavior

There is to communication with tft or pronterface.

### Steps to Reproduce

1. Download marlil 2.1.2.2 with preconfigured files.
2. Uncomment Input shaping
3. Compile and upload to the printer.

### Version of Marlin Firmware

2.1.2.2

### Printer model

Artillery Sidewinder X2

### Electronics

Stock Artillery rubyboard

### LCD/Controller

Stock LCD Controller

### Other add-ons

All electronic parts are stock

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/14300932/Marlin.zip)
Processing Artillery_Ruby (board: marlin_Artillery_Ruby; platform: ststm32@~12.1; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Copying variant MARLIN_ARTILLERY_RUBY to framework directory...
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_Artillery_Ruby.html
[... 146 lines cut ...]
Compiling .pio/build/Artillery_Ruby/src/src/module/servo.cpp.o
Compiling .pio/build/Artillery_Ruby/src/src/module/settings.cpp.o
Compiling .pio/build/Artillery_Ruby/src/src/module/stepper.cpp.o
Compiling .pio/build/Artillery_Ruby/src/src/module/stepper/indirection.cpp.o
Compiling .pio/build/Artillery_Ruby/src/src/module/temperature.cpp.o
Compiling .pio/build/Artillery_Ruby/src/src/module/tool_change.cpp.o
Compiling .pio/build/Artillery_Ruby/src/src/sd/Sd2Card.cpp.o
Compiling .pio/build/Artillery_Ruby/src/src/sd/SdBaseFile.cpp.o
Compiling .pio/build/Artillery_Ruby/src/src/sd/SdFatUtil.cpp.o
Compiling .pio/build/Artillery_Ruby/src/src/sd/SdFile.cpp.o
Compiling .pio/build/Artillery_Ruby/src/src/sd/SdVolume.cpp.o
Compiling .pio/build/Artillery_Ruby/src/src/sd/cardreader.cpp.o
Linking .pio/build/Artillery_Ruby/firmware.elf
Checking size .pio/build/Artillery_Ruby/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  44.3% (used 29056 bytes from 65536 bytes)
Flash: [=====     ]  47.8% (used 125336 bytes from 262144 bytes)
Building .pio/build/Artillery_Ruby/firmware.bin
========================================================== [SUCCESS] Took 238.84 seconds ==========================================================

Environment     Status    Duration
--------------  --------  ------------
Artillery_Ruby  SUCCESS   00:03:58.836
=========================================================== 1 succeeded in 00:03:58.836 ===========================================================
 *  Terminal will be reused by tasks, press any key to close it.

thisiskeithb, 2024-02-15:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

WenzelHyde, 2024-02-15:
I already tried, I thought that is not compatibile with 2.1.2.2 because of result:

Processing Artillery_Ruby (board: marlin_Artillery_Ruby; platform: ststm32@~12.1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:39,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:284:11: error: missing binary operator before token "("
  284 | #if EITHER(PARKING_EXTRUDER, MAGNETIC_PARKING_EXTRUDER)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:986:11: error: missing binary operator before token "("
  986 | #if EITHER(MORGAN_SCARA, MP_SCARA)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:1787:11: error: missing binary operator before token "("
 1787 | #if EITHER(MIN_SOFTWARE_ENDSTOPS, MAX_SOFTWARE_ENDSTOPS)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:1989:11: error: missing binary operator before token "("
 1989 | #if EITHER(AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_BILINEAR)
      |           ^
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:39,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:3035:11: error: missing binary operator before token "("
 3035 | #if EITHER(ANYCUBIC_LCD_I3MEGA, ANYCUBIC_LCD_CHIRON)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:3323:11: error: missing binary operator before token "("
 3323 | #if EITHER(RGB_LED, RGBW_LED)
      |           ^
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:56,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:305:11: error: missing binary operator before token "("
  305 |   #if BOTH(ADAPTIVE_FAN_SLOWING, PIDTEMP)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1089:11: error: missing binary operator before token "("
 1089 | #if EITHER(INPUT_SHAPING_X, INPUT_SHAPING_Y)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1325:11: error: missing binary operator before token "("
 1325 | #if EITHER(DIGIPOT_MCP4018, DIGIPOT_MCP4451)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1382:28: error: missing binary operator before token "("
 1382 | #if HAS_BED_PROBE && EITHER(HAS_MARLINUI_MENU, HAS_TFT_LVGL_UI)
      |                            ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1437:11: error: missing binary operator before token "("
 1437 | #if EITHER(HAS_DISPLAY, DWIN_LCD_PROUI)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1501:26: error: missing binary operator before token "("
 1501 | #if HAS_DISPLAY && EITHER(SDSUPPORT, SET_PROGRESS_MANUALLY)
      |                          ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2149:11: error: missing binary operator before token "("
 2149 | #if EITHER(AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_UBL)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2184:11: error: missing binary operator before token "("
 2184 | #if EITHER(MESH_BED_LEVELING, AUTO_BED_LEVELING_UBL)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2192:9: error: missing binary operator before token "("
 2192 | #if BOTH(AUTO_BED_LEVELING_UBL, EEPROM_SETTINGS)
      |         ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2306:11: error: missing binary operator before token "("
 2306 | #if EITHER(ARC_SUPPORT, BEZIER_CURVE_SUPPORT)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2394:9: error: missing binary operator before token "("
 2394 | #if BOTH(SDSUPPORT, DIRECT_STEPPING)
      |         ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:3336:11: error: missing binary operator before token "("
 3336 | #if EITHER(SPINDLE_FEATURE, LASER_FEATURE)
      |           ^
Error: Failed to parse Marlin features. See previous error messages.
=============================================================== [FAILED] Took 14.05 seconds ===============================================================

Environment     Status    Duration
--------------  --------  ------------
Artillery_Ruby  FAILED    00:00:14.052
========================================================== 1 failed, 0 succeeded in 00:00:14.052 ==========================================================

 *  The terminal process "platformio 'run'" terminated with exit code: 1.

ellensp, 2024-02-15:
You CANNOT mix config files and code versions, you must use matching versions

Bugfix configs do not include EITHER or BOTH macros.

ellensp, 2024-02-15:
here are the bugfix Config files for Artillery Sidewinder X2
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Artillery/Sidewinder%20X2 
Part of this archive https://github.com/MarlinFirmware/Configurations/archive/refs/heads/bugfix-2.1.x.zip

WenzelHyde, 2024-02-16:
Sorry, I'm trying my best, but with linked files I'm getting only this... I tried increase editor.maxTokenizationLineLength, still the same. After compile attempt I'M getting error: "MOTHERBOARD not defined! Use '#define MOTHERBOARD BOARD_...' in Configuration.h." But in Configuration.h is "#define MOTHERBOARD ARTILLERY_RUBY"
[image]

WenzelHyde, 2024-02-16:
Processing Artillery_Ruby (board: marlin_Artillery_Ruby; platform: ststm32@~12.1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:36,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins.h:1026:6: error: #error "MOTHERBOARD not defined! Use '#define MOTHERBOARD BOARD_...' in Configuration.h."
 1026 |     #error "MOTHERBOARD not defined! Use '#define MOTHERBOARD BOARD_...' in Configuration.h."
      |      ^~~~~
Error: Failed to parse Marlin features. See previous error messages.
==================================================================== [FAILED] Took 6.18 seconds ====================================================================

Environment     Status    Duration
--------------  --------  ------------
Artillery_Ruby  FAILED    00:00:06.175
=============================================================== 1 failed, 0 succeeded in 00:00:06.175 ===============================================================

ellensp, 2024-02-16:
You can do that that but you must click raw first, so it shows you the raw file to download.

WenzelHyde, 2024-02-17:
Guys, thank you for your patience. I guess I'm too old for tech... 

Now it is working as expected. Just for curiosity, is it repaired in bugfix version, or was something removed from 2.1.2.2?

Anyway, thank you.

oaragues, 2024-03-30:
> Guys, thank you for your patience. I guess I'm too old for tech...
> 
> Now it is working as expected. Just for curiosity, is it repaired in bugfix version, or was something removed from 2.1.2.2?
> 
> Anyway, thank you.

So, how did you fixed it? I'm having the same issue, when undashing the #define input_shaping the printer can't connect with Pronterface, did you download the configuration bugfix?

thisiskeithb, 2024-03-30:
Use the latest `bugfix-2.1.x`. Input Shaping causes bootloops on `2.1.2.2`.

closed 2024-03-30 by thisiskeithb