MarlinFirmware/Marlin #26798: [BUG] Power Loss Resume not working

buff0k, 2024-02-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Running an Ender-5 Pro with CR Touch, prior to installing the CR Touch, Power Loss Resume worked as expected, however since installing the CR Touch, this feature no longer works.
Manually triggering power failure (Pulling the Cord) and then reading the SD Card shows that the Resume file is not being writte to the SD Card.

### Bug Timeline

New, since installing CR Touch

### Expected behavior

In the case of power loss, Marlin looks for the Resume file at startup and, if found, prompts user to resume print.

### Actual behavior

In the case of power loss, Marlin boots up normally, no Resume file on SD Card

### Steps to Reproduce

1. Start Print
2. Simulated/Actual Power Loss to Printer
3. Printer Starts up, does not have option to resume

### Version of Marlin Firmware

2.1.2.1 and 2.1.x-bugfix

### Printer model

Ender-5 Pro

### Electronics

Stock V4.2.2 Motherboard

### LCD/Controller

Stock Ender-5 Pro

### Other add-ons

CR Touch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[2.1.2.1.zip](https://github.com/MarlinFirmware/Marlin/files/14318053/2.1.2.1.zip)
[2.1.x-bugfix.zip](https://github.com/MarlinFirmware/Marlin/files/14318055/2.1.x-bugfix.zip)

ellensp, 2024-02-17:
"#define PLR_ENABLED_DEFAULT       false // Power-Loss Recovery enabled by default. (Set with 'M413 Sn' & M500)"

did you actually enable it?

buff0k, 2024-02-17:
I did, I set to false in configuration.h to see if that wasn't causing the issue (commit to EEPROM), but even with it set to true or enabled manually (with or without commit- or power cycling) it happens.
Updated issue (Likely unrelated bug) is that 2.1.2.2 doesn't respect Z-Probe endstop, but for now still testing on 2.1.2.1 and 2.1.x-bugfix to resolve the power loss issue.

closed 2024-06-02 by github-actions[bot]