MarlinFirmware/Marlin #26799: [BUG] (printer restarts on and on)

mohammad787, 2024-02-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

after updating to the latest firmware 2.1.2.2 the printer works fine just one time and after powering off and on the printer starts behaving crazy and restarts every one to five minutes

### Bug Timeline

It happened right away after updating

### Steps to Reproduce

powering off and on the power supply

### Version of Marlin Firmware

2.1.2.2

### Printer model

SKR2 V2 ON ENDER 3 DUAL Z WITH INDEPENDENT Z STEPPER 

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

mohammad787, 2024-02-18:
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/14321419/configs.zip)

thisiskeithb, 2024-02-18:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

mohammad787, 2024-02-18:
downloaded it but could not compile. it gives me too many errors like :
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:287:11: error: missing binary operator before token "(" 
  287 | #if EITHER(PARKING_EXTRUDER, MAGNETIC_PARKING_EXTRUDER)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:989:11: error: missing binary operator before token "(" 
  989 | #if EITHER(MORGAN_SCARA, MP_SCARA)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:1800:11: error: missing binary operator before token "("
 1800 | #if EITHER(MIN_SOFTWARE_ENDSTOPS, MAX_SOFTWARE_ENDSTOPS)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:2005:11: error: missing binary operator before token "("
 2005 | #if EITHER(AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_BILINEAR)
      |           ^
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:39,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:3068:11: error: missing binary operator before token "("
 3068 | #if EITHER(ANYCUBIC_LCD_I3MEGA, ANYCUBIC_LCD_CHIRON)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:3356:11: error: missing binary operator before token "("
 3356 | #if EITHER(RGB_LED, RGBW_LED)
      |           ^
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:56,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:305:11: error: missing binary operator before token "("
  305 |   #if BOTH(ADAPTIVE_FAN_SLOWING, PIDTEMP)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1090:11: error: missing binary operator before token "("
 1090 | #if EITHER(INPUT_SHAPING_X, INPUT_SHAPING_Y)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1326:11: error: missing binary operator before token "("
 1326 | #if EITHER(DIGIPOT_MCP4018, DIGIPOT_MCP4451)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1383:28: error: missing binary operator before token "("
 1383 | #if HAS_BED_PROBE && EITHER(HAS_MARLINUI_MENU, HAS_TFT_LVGL_UI)
      |                            ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1438:11: error: missing binary operator before token "("
 1438 | #if EITHER(HAS_DISPLAY, DWIN_LCD_PROUI)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1496:11: error: missing binary operator before token "("
 1496 |   #if BOTH(M73_REPORT, SDSUPPORT)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1502:26: error: missing binary operator before token "("
 1502 | #if HAS_DISPLAY && EITHER(SDSUPPORT, SET_PROGRESS_MANUALLY)
      |                          ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2151:11: error: missing binary operator before token "("
 2151 | #if EITHER(AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_UBL)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2186:11: error: missing binary operator before token "("
 2186 | #if EITHER(MESH_BED_LEVELING, AUTO_BED_LEVELING_UBL)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2194:9: error: missing binary operator before token "("
 2194 | #if BOTH(AUTO_BED_LEVELING_UBL, EEPROM_SETTINGS)
      |         ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2308:11: error: missing binary operator before token "("
 2308 | #if EITHER(ARC_SUPPORT, BEZIER_CURVE_SUPPORT)
      |           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2396:9: error: missing binary operator before token "("
 2396 | #if BOTH(SDSUPPORT, DIRECT_STEPPING)
      |         ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:3176:13: error: missing binary operator before token "("
 3176 |   #if EITHER(SENSORLESS_HOMING, SENSORLESS_PROBING)
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:3338:11: error: missing binary operator before token "("
 3338 | #if EITHER(SPINDLE_FEATURE, LASER_FEATURE)
      |           ^
Error: Failed to parse Marlin features. See previous error messages.

ellensp, 2024-02-18:
You MUST use matching Config files and code.

Current bugfix config files do not contain EITHER or BOTH Macros

Update the configuration files to bufix2.1.x standard

mohammad787, 2024-02-18:
Yup the bugfix version fixed the problem 
thank you.

mohammad787, 2024-02-20:
ok I tested it and after just the first print the exact problem kicked in and...
here is the modified config files
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/14340016/Configuration.zip)

mohammad787, 2024-02-20:
thanks for the fast reply
1.the mainboard is skr2 which the most compatible config file that I found was biqu/b1
then ok I'll remove the direction
2.#define BABYSTEP_INVERT_Z false              // Enable if Z babysteps should go the other way
yeah didn't see that I really add false for no reason
3.and the back margin meant to have a different value to be tested but then I came to the same value as the other sides "I'll correct it too"
4.microstepping was also to check the highest value to be handled by printer "I'll reduce them as well"

mohammad787, 2024-02-20:
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/14343915/configs.zip)
I did all of that still the same

mohammad787, 2024-02-20:
FOUND THE PROBLEM 
I set a timer to find out the intervals and it was exactly 3 minutes 
so searched here and it sounds skr2 board has a problem with SD card after updates and it must be removed or it restarts every 3 minutes!!!

thank you guys anyway

closed 2024-08-29 by github-actions[bot]