MarlinFirmware/Marlin #26800: [BUG] (ENDER-5-plus - Pause/Resume and Filament Sensor not working correctly)

JoJos1220, 2024-02-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi Community,

I got an issue with the Creality Ender-5-plus.
Currently I am updating the 3D-Printer with custom BigTreetech SKR Mini v3.0 Mainboard. The printing is done correctly, but the Pause/Resume function and the Filament Runout Function does not work correctly.
The Printhead is successfully moving into the Parking-Position which is confirmed by a display Popup says "Parikg Position" Abort/Resume. After hitting "Resume" on the Display - nothing is happening and the Printer stays in the parking position forever until a power cut.

### Bug Timeline

Always present until test

### Expected behavior

Pause/Resume should work as excepted.
After Filament Runout - Filament Change - Printer should going on with print.

### Actual behavior

Printer stopps in parking position and stays forever until power cut

### Steps to Reproduce

1. Load Marlin 2.1.2.2 OR Marlin 2.1.x bugfix Firmware on Ender-5-plus with BTT SKR Mini v3 Mainboard
2. Load LCD DGUS-Reloaded FW on Stock Display
3. Start a default print
4. Select Pause or Pull Out filament on Filament sensor
5. Wait until Parking Position has reached
6. Hit Resume
7. Printer is in Error mode
8. [Config.zip](https://github.com/MarlinFirmware/Marlin/files/14322254/Config.zip)

### Version of Marlin Firmware

2.1.2.2 OR 2.1.x Bugfix

### Printer model

Creality Ender-5-plus

### Electronics

Stock LCD, BTT SKR Mini v3.0 Mainboard

### LCD/Controller

STOCK LCD Creality Ender-5-plus T5 dwon.com.cn N42C21.00HG08 2126

### Other add-ons

Direct Drive (Should not matter....)

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ursius2020, 2024-08-03:
Can confirm this bug, also present on Silent Board V2.2.1

JoJos1220, 2024-08-04:
> Can confirm this bug, also present on Silent Board V2.2.1

Yeah?

I also retested it about 1-2 Month ago with the latest Bugfix Version again - and it is still present to me.....

ursius2020, 2024-08-04:
Yes, tested it the past days. Present on latest Bugfix as well es 2.1.2.4

YannickAdriaenssens, 2024-09-04:
Having the same issue on BTT SKR3 on ender 5 plus with bugfix-2.1.x

YannickAdriaenssens, 2025-04-23:
Hi, did someone find a solution to this problem and can help me out? This is the last bug which prevent me from using my ender 5 plus with stock LCD and SKR3 board efficiently. At this point all my prints fail when filament runs out. I'm collecting a lot of spools with a little filament which I'm scared to use if it would run out...

If someone could help, I would highly appreciate it! 

Thanks 
Yannick

ThomasToka, 2025-04-23:
Hi,
you could try if my fix for the Ender 3 S1 Pro in my fork works for this also:

https://github.com/ThomasToka/Marlin/blob/1a53db36a1d930770b239cd9fbd32503f40ebf49/Marlin/src/feature/pause.cpp#L803

If you dont want to guard it on compile time simply add this in the line:
```
pause_menu_response = PAUSE_RESPONSE_WAIT_FOR;
```

closed 2026-10-29 by thinkyhead