MarlinFirmware/Marlin #26801: "#define Y2_DRIVER_TYPE xxxx" but "Y2_SPEP_PIN" not work

SeanW2024, 2024-02-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

V2.1.2&V2.1.2.1 works well.In version 2.1.2.2, the corresponding pins did not generate pulse signals.In fact, if E1_STEP_PIN is defined in pin. h, when # define X2-DRIVER-TYPE A4988, compilation will prompt "Note: Auto assigned X2 STEP/DIR/ENABLE-PINs to unused En_STEP/DIR/ENABLE-PINs." However, when # define Y2-DRIVER-TYPE A4988,the compiler will not provide any prompts.

 

### Version of Marlin Firmware

2.1.2.2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-02-18:
In bugfix HAS_DUAL_Y_STEPPERS was removed

but 2.1.2.2 still uses it, but this was omitted from thr 2.1.2.2 code

open up Marlin/src/inc/Conditionals_LCD.h

Find 
```
#ifdef Y2_DRIVER_TYPE
  #define HAS_Y2_STEPPER 1
#endif
```
replace it with
```
#ifdef Y2_DRIVER_TYPE
  #define HAS_Y2_STEPPER 1
  #define HAS_DUAL_Y_STEPPERS 1
#endif

```

SeanW2024, 2024-02-19:
Thank you. Now,it works properly

thisiskeithb, 2024-03-28:
Closing since this is already fixed in `bugfix-2.1.x`.

closed 2024-03-28 by thisiskeithb

thisiskeithb, 2024-04-28:
Patch for `2.1.x` submitted:
- https://github.com/MarlinFirmware/Marlin/pull/27024