MarlinFirmware/Marlin #26802: [BUG] (temperature.h lacks Temperature::preheat_end_ms_hotend in official Marlin 2.1.2.2 download zip package)

Fiamat, 2024-02-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After setting MILLISECONDS_PREHEAT_TIME to any positive value the temperature.cpp checks for nonexisting member preheat_end_ms_hotend in Temperature class. Indeed temperature.h doesn't declare such variable. I see that bugfix branch already brought back this variable, but official .zip with Marlin 2.1.2.2 (https://marlinfw.org/meta/download/) contains this bug.

### Actual behavior

During build process a following error is thrown:

`Marlin\src\module\temperature.cpp:598:12: error: 'millis_t Temperature::preheat_end_ms_hotend [1]' is not a static data member of 'class Temperature'`

### Steps to Reproduce

1. Uncomment `MILLISECONDS_PREHEAT_TIME  ` in `Configuration_adv.h` and set it to any positive value
2. Try building the software

My Configuration(_adv).h:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/14323205/Marlin.zip)

### Version of Marlin Firmware

2.1.2.1

### Printer model

Ender 3 v2

### Electronics

Bigtreetech SKR Mini E3 v3

### LCD/Controller

StockEnder 3 v2

### Other add-ons

BL-Touch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-02-19:
As a work around In Marlin/src/module/temperature.cpp

Find 
```CPP
#if MILLISECONDS_PREHEAT_TIME > 0
  millis_t Temperature::preheat_end_ms_hotend[HOTENDS]; // = { 0 };
#endif
```
and replace it with 
```CPP
#if MILLISECONDS_PREHEAT_TIME > 0
  millis_t Temperature::preheat_end_time[HOTENDS] = { 0 };
#endif
```

Fiamat, 2024-02-19:
> As a work around In Marlin/src/module/temperature.cpp
> 
> Find
> 
> ```c++
> #if MILLISECONDS_PREHEAT_TIME > 0
>   millis_t Temperature::preheat_end_ms_hotend[HOTENDS]; // = { 0 };
> #endif
> ```
> 
> and replace it with
> 
> ```c++
> #if MILLISECONDS_PREHEAT_TIME > 0
>   millis_t Temperature::preheat_end_time[HOTENDS] = { 0 };
> #endif
> ```

I use temperature sensor "66 : Trianglelab T-D500 500°C High Temperature Thermistor" which has a long dead zone around room temperature. Event the Marlin itself throws warning when MILLISECONDS_PREHEAT_TIME is below 30e3 ms with this sensor. I guess that this workaround can cause problems with temperature control (?).

ellensp, 2024-02-19:
Marlin has this code 
```
#if ANY_E_SENSOR_IS(66)
...(skipped code)... 
 #if MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED < 5
    #error "Thermistor 66 requires MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED ≥ 5."
  #elif MILLISECONDS_PREHEAT_TIME < 15000
    #error "Thermistor 66 requires MILLISECONDS_PREHEAT_TIME ≥ 15000, but 30000 or higher is recommended."
```

to ensure MILLISECONDS_PREHEAT_TIME is > 15000

thisiskeithb, 2024-03-28:
Closing since this is already fixed in `bugfix-2.1.x`.

closed 2024-03-28 by thisiskeithb

thisiskeithb, 2024-04-28:
Patch for `2.1.x` submitted:
- https://github.com/MarlinFirmware/Marlin/pull/27023