MarlinFirmware/Marlin #26807: [BUG] Homing fail causes print issues on start

classicrocker883, 2024-02-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

- (I'm using UBL, w/ CR-Touch bed probe + ProUI on Ender-3V2 style printer)

When I start a print, and say homing fails (for the probe/Z axis), it does an odd behavior.  
It seems to move with no speed/feedrate. but more importantly it prints in midair at the Z axis level where it failed to Home Z.

If there was a way to retry Homing the Z axis, and if it still fails, to abort the print.

### Expected behavior

After the Z axis fails to home/be trusted (w/ or w/o a probe), it should retry _homing_, and probably abort the print after so many tries.

### Actual behavior

After say the probe fails to deploy, it still proceeds to continue the print, except way above the bed surface, and at a very slow speed.

### Steps to Reproduce

1. Start a print.
2. Somehow cause the probe or endstop to fail for the Z axis.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Voxelab Aquila

### Electronics

CR-Touch

### LCD/Controller

ProUI

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

GMagician, 2024-03-03:
Are you printing with SD? I think I also saw this behaviour but I had no occasion to analyze it

classicrocker883, 2024-03-05:
> Are you printing with SD? I think I also saw this behaviour but I had no occasion to analyze it. I'm using BL-Touch as probe

yes using SD. I have also used Octoprint / pi and it does the same. Of course a normal working probe won't give any issue, but I was having an issue with the probe being "sticky" or hard to move the pin up and down. which resulted in as stated above: homing fails, yet it continues to print - on the wrong level/height.

I think a good fix would be something like [not exactly verbatim, but to give an idea]:    
```y
if (homing_fail) {homing_fail++; try home_again; }
else if (homing_fail > 2 (times)) { return LCD_message(MSG_KILL_HOMING_FAILED); }  // "Homing Failed"
```

GMagician, 2024-03-05:
The problem here is that probe may fail in multiple parts. I saw this during G34 but everywhere the probe is used you may have issues. What I noticed with my BLTouch is that when it fails it does a multiple "deploy and stow" action before ending in "nothing".

Home and G34 fail may be crucial when they occur during print so LCD_message is not enough to stop printing. Do it when using SD should be easier, not sure when printing via host (if you don't have HOST_ACTION_COMMANDS and/or HOST_PROMPT_SUPPORT)

thisiskeithb, 2024-03-05:
>  but more importantly it prints in midair at the Z axis level where it failed to Home Z.

This is probably the best case scenario of a failed `G28`. The types of failed homing attempts I see are ones where you _want Marlin to halt_ because further attempts will cause damage.

GMagician, 2024-03-05:
The failure is something that should not occur, not sure that a retry action is the correct solution. I was adding a fatal halt to all probe failure

thisiskeithb, 2024-03-05:
> Of course a normal working probe won't give any issue, but I was having an issue with the probe being "sticky" or hard to move the pin up and down. which resulted in as stated above: homing fails, yet it continues to print - on the wrong level/height.

That sounds like a probe failing and triggering as homed. Marlin doesn’t know the difference between a “good” home or a “bad” home, so of course it’s going to continue the print job as if it had successfully homed.

Fix or replace the probe.

closed 2024-03-05 by thisiskeithb

GMagician, 2024-03-06:
@thisiskeithb are you really sure Marlin can't be aware of that?
During G34 when probe fails, I see that my bl-touch deploy and stow multiple times without moving X then it ends. Code return true when probe fail. I think that a fatal error should be sent, if printer is printing, otherwise a simple status message

thisiskeithb, 2024-03-06:
> are you really sure Marlin can't be aware of that?

If a probe reports a failure, sure. But if a probe's failure mode is to simply trigger as normal (at the wrong height), then Marlin won't know the difference.

GMagician, 2024-03-06:
> > are you really sure Marlin can't be aware of that?
> 
> If a probe reports a failure, sure. But if a probe's failure mode is to simply trigger as normal (at the wrong height), then Marlin won't know the difference.

What I see in my specific case is not a height failure but a probe failure. If this kind of failure occurs while homing you will probably end with a print in air. I changed code to halt on it but, as often happens, it's now working with no failure