MarlinFirmware/Marlin #26808: [BUG] Nonlinear extrusion - blob at the line srart

vovodroid, 2024-02-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

There is big overextrusion at the line start with nonlinear extrusion enabled

[image]

Are NLE applied for unretractions? According to feature description `Only adjusts forward extrusions, since those are the ones affected by backpressure.`

I searched in the code, but the only check I found is https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/module/stepper.cpp#L2742

```
        if (current_block->direction_bits.e) {
          ne_fix.A = (1L << 24) * ne.A;
          ne_fix.B = (1L << 24) * ne.B;
          ne_fix.C = (1L << 24) * ne.C;
        }
        else {
          ne_fix.A = ne_fix.B = 0;
          ne_fix.C = (1L << 24);
        }
```

As comment says `Direction bits set for this block, where 1 is negative motion`, so will factors be applied for unretract in this case (most chances I miss something, because blob is on unretract, e.g. forward extrusion)? 

Also is correction applied for E only move (unretract case) too? If so, it should be fixed, retracts has very big speed, so correction is also very large, producing such blobs.

### Version of Marlin Firmware

Latest bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

vovodroid, 2024-02-21:
Will 

`if (current_block->direction_bits.e && !axis_did_move.x && axis_did_move.y)`

be correct fix?

vovodroid, 2024-02-21:
@abortz, could you take a look please?

abortz, 2024-02-23:
Thanks for the heads up @vovodroid! Let me do some testing, but I think you're right that this can be a problem. I believe I may have noticed some line start blobbing myself, but it's always been so minor I've written it off.

I'm kinda hesitant to disable NLE on unretracts, since the core problem we're trying to solve exists regardless of how the extruder is being used, and unretracts are typically done at very high speeds where the problem is exacerbated. I was hoping that no NLE on retracts and NLE on unretracts would be "correct" in a sense, but perhaps not... Obviously it's far more important to get retraction dead-on right!

vovodroid, 2024-02-23:
> it's always been so minor

By mistake I swapped A and B coefficients (wrong mathematics order is very confusing, it's RepRap legacy, I guess), so result was much higher for high speed.

> unretracts are typically done at very high speeds

Yes, but most of unretract path is before orifice itself. Imagine Bowed extractor with retract 5 mm, at high unrectact speed you can easily arrive to 7-10 mm unrectact, resulting in huge blob. Probably it could be solved by applying correction to final 0.2-0.3 mm, but it's too complicated. Better is to set unretract extra length in slicer/Marlin.

abortz, 2024-02-23:
> Yes, but most of unretract path is before orifice itself. Imagine Bowed extractor with retract 5 mm, at high unrectact speed you can easily arrive to 7-10 mm unrectact, resulting in huge blob. [...]

That is a great point! So then, since having Marlin know when the filament is and isn't actually *pushed* into the bottleneck is way too much to expect, I'm pretty convinced then disabling NLE for unretracts is the best option.

In the planner Marlin checks for retracts/unretracts here: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/module/planner.cpp#L2466
That's probably how we should do it as well, if you want to give it another stab. Haven't done any Marlin development in a while, need time to spin up the dev environment again.

vovodroid, 2024-02-23:
> In the planner Marlin checks for retracts/unretracts here: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/module/planner.cpp#L2466

But NLE is applied in interrupt service routine latter. Actually planner code check axis steps, that's what I tried in ISR, but it doesn't work in ISR.

abortz, 2024-02-23:
> > In the planner Marlin checks for retracts/unretracts here: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/module/planner.cpp#L2466
> 
> But NLE is applied in interrupt service routine latter. Actually planner code checks axis steps, that's what I tried in ISR, but it doesn't work in ISR.

I know NLE is applied later, but I don't see why it didn't/wouldn't work the same in the scheduler there, the step count is definitely populated there.

vovodroid, 2024-02-23:
Would you like to move NLE from stepper ISR to planner? Could SERIAL debug be used there?

abortz, 2024-02-24:
Well at least part needs to live in stepper interrupts, because it needs access to the *current* feedrate factoring in accelerations etc. I hadn't thought about the per-block setup living in the planner, and sadly I have no experience with Marlin serial debug. If you're having trouble making it work, I'll spin up an environment and try my hand.

vovodroid, 2024-02-24:
> I have no experience with Marlin serial debug.

I mean just output log to serial port, but it seems doesn't work in ISR.

Actually my question why `if (current_block->direction_bits.e && !axis_did_move.a & axis_did_move.b)` didn't help?

Code you referred in planner

```
if (true NUM_AXIS_GANG(
      && !block->steps.a, && !block->steps.b, && !block->steps.c,
      && !block->steps.i, && !block->steps.j, && !block->steps.k,
      && !block->steps.u, && !block->steps.v, && !block->steps.w)
  ) {                                                             // Is this a retract / recover move?
    accel = CEIL(settings.retract_acceleration * steps_per_mm);   // Convert to: acceleration steps/sec^2
  }
```

checks the same.

abortz, 2024-02-26:
> Actually my question why `if (current_block->direction_bits.e && current_block->steps.a == 0 && current_block->steps.b == 0)` didn't help?

Oh, I think you've got the logic reversed. We want to apply NLE if: we are extruding forward AND we are moving any axis. Try:
`if (current_block->direction_bits.e && (current_block->steps.a || current_block->steps.b))`

vovodroid, 2024-02-26:
> I think you've got the logic reversed.

Just dummy concentration, is there facepalm smile?))) Sure you are right, it's working now!

What is the correct way to test it?

NUM_AXIS_GANG with `block->steps.*` or `didmove.*`?

abortz, 2024-02-26:
> > I think you've got the logic reversed.
> 
> Just dummy concentration, is there facepalm smile?))) Sure you are right, it's working now!

Yay!!!

> What is the correct way to test it?
> 
> NUM_AXIS_GANG with `block->steps.*` or `didmove.*`?

*shrug*

Submit a pull request and we can continue the conversation there and get some more eyes on the code. For all intents and purposes what you have there is probably fine, but the "correct" way is the way it's done in the planner, since that would work with all motion systems including unusual ones.

thisiskeithb, 2024-02-27:
Closing since you’ve opened a PR.

PR #26824 (2024-04-07): Disable nonlinear extrusion on unretract
Resolves #26808

Don't use nonlinear extraction on untertracts to prevent blobs at the line start, use only when any axis is moving.
closed 2024-02-27 by thisiskeithb