MarlinFirmware/Marlin #26809: [BUG] Auto-home on v.2.1.2.2 crashes nozzle into bed w/CR Touch

jradwan, 2024-02-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After upgrading to 2.1.2.1 to 2.1.2.2 (and -bugfix), when I perform an auto-home on my Ender 3 Pro with a CR Touch installed, the probe extends and the Z-axis starts moving downward. But after the probe touches the bed the Z-axis continues to move down and grinds the stepper motor and nozzle into the bed until I turn the power off. 

Going back to 2.1.2.1 fixes the issue.

### Bug Timeline

after upgrading from 2.1.2.1 to 2.1.2.2

### Expected behavior

CR Touch/Z-probe should stop nozzle from crashing into bed

### Actual behavior

Probe touches bed, light turns red, but Z-axis continues to move down grinding stepper motor and driving nozzle into bed until power is shut off.

### Steps to Reproduce

Motion > Auto Home

### Version of Marlin Firmware

2.1.2.bugfix

### Printer model

Ender 3 Pro

### Electronics

4.2.7 32-bit board

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

All of my changes/customizations are tagged with "JCR"  in these files. Attached are my working files from 2.1.2.1 and the files used with 2.1.2.x-bugfix that cause the problem.

[2-1-2-x-bugfix-configs-auto-home-crash.zip](https://github.com/MarlinFirmware/Marlin/files/14377040/2-1-2-x-bugfix-configs-auto-home-crash.zip)

ellensp, 2024-02-23:
What type of MPU is on your creality 4.2.7 STM32F103RE, STM32F103RC or GD32F303RE ?
And what build environment did you use?

tu77, 2024-02-23:
after Upgrading from 2.1.2.1 to 2.1.2.2 my double Z-Axis is not homing anymore, I get a "kill" after some minutes, kind of timeout..
maybe it is a similar issue.

I did compare my old and new config files but find no reason for the Problem. 

Printer type:
core XY 
dual Z axis
2 Z-home switches only (no BL or probe)

jradwan, 2024-02-23:
> What type of MPU is on your creality 4.2.7 STM32F103RE, STM32F103RC or GD32F303RE ? 

STM32F103RET6

> And what build environment did you use?

default_envs = STM32F103RET6_creality

> And where is your probe attached? from the config the probe should be using the 5 pin probe port and nothing else

Yes, attached to the dedicated probe port on the board, original Z-stop removed nothing plugged in there.

classicrocker883, 2024-02-25:
i also have CR-touch with 427, I have never noticed this issue. I have heard there may be issues with this and HS (high speed) mode and definitely had experienced problems with this enabled like using M48.

what UI are you using, ProUI? Jyers?

jradwan, 2024-02-25:
I have not enabled BL_TOUCH_HS_MODE and Z_MIN_PROBE_REPEATABILITY_TEST is also disabled.

I'm on an Ender 3 Pro, so no special UI, just the stock LCD and rotary encoder.

stonesword0, 2024-02-26:
I had this same issue with the new firmware. I have a BLTouch, so it is likely an issue with the 4.2.7 mainboard.

Thank you for reporting this, I would have never figured out why my BLTouch isn't working!

jradwan, 2024-02-26:
> Thank you for reporting this, I would have never figured out why my BLTouch isn't working!
I'm glad I'm not the only one!

buff0k, 2024-02-26:
Can confirm same issue with 4.2.2 Motherboard on an Ender-5 Pro.

ellensp, 2024-02-26:
Me two-ing provides extremely limited information.

Provide Config files and details of how your bltouch is wired (probe port, probe port and z-stop, or using an adapter such as a pin 27 adapter) And what probe you really have bltouch, crtouch, 3dtouch etc

ellensp, 2024-02-27:
I think i've found something which looks like a very likely suspect to break any probe n the probe port.

In 2.1.2.2 Marlin/src/inc/Conditionals_post.h is
```CPP
 #if ALL(ENDSTOPPULLUPS, USE_Z_MIN_PROBE)
    #define ENDSTOPPULLUP_ZMIN_PROBE
  #endif
```
But USE_Z_MIN_PROBE is not part of 2.1.2.2, it came from bugfix by mistake.
So ENDSTOPPULLUP_ZMIN_PROBE is never defined.

Update this code to
```CPP
  #if ALL(ENDSTOPPULLUPS, HAS_Z_MIN_PROBE_PIN)
    #define ENDSTOPPULLUP_ZMIN_PROBE
  #endif
```
And ENDSTOPPULLUP_ZMIN_PROBE gets defined.

Can anyone with this hardware setup very carefully test this?

jradwan, 2024-02-27:
The patch did not work for me. Was that for the bugfix-2.1.x branch? Or should I have applied it to 2.1.2.2?

jradwan, 2024-02-27:
Ok, maybe I'm confused then because I was having the issue with bugfix-2.1.x as well as 2.1.2.2.

But I just tried the proposed patch on the 2.1.x branch (2.1.2.2 with bugfixes) and it worked, the auto-home completes successfully.

ellensp, 2024-02-27:
2.1.x and 2.1.2.2 are currently identical.
branch bugfix-2.1.x. is what we refer to as bugfix not branch 2.1.x

jradwan, 2024-02-27:
Ok, I must have downloaded something wrong or messed up on my side then when filing the bug originally because I thought I _was_ able to reproduce on bugfix-2.1.x. But whatever, the patch works for 2.1.2.2 so I'm back in business. Thanks!

buff0k, 2024-02-27:
Can confirm above fix does correct the issue on 2.1.2.2 on Ender-5 Pro with 4.2.2 board and CR Touch.
Thank you @ellensp

classicrocker883, 2024-02-27:
@ellensp  
>And BLTOUCH_HS_MODE and BLTOUCH_FORCE_SW_MODE cannot be used on crtouch or 3dtouch.

im wondering why cant FORCE_SW_MODE be used? even though it says: `and most clones:`
```r
  /**
   * Settings for BLTOUCH Classic 1.2, 1.3 or BLTouch Smart 1.0, 2.0, 2.2, 3.0, 3.1, and most clones:
   */

  // Feature: Switch into SW mode after a deploy. It makes the output pulse longer. Can be useful
  //          in special cases, like noisy or filtered input configurations.
  //#define BLTOUCH_FORCE_SW_MODE
```

thisiskeithb, 2024-02-27:
> im wondering why cant FORCE_SW_MODE be used? even though it says: `and most clones:`

Because "most clones" referred to the "3D Touch" clones that existed long before Creality's CRTouch.

gitana2, 2024-02-29:
I confirm on Ender 3S1 (stock firmw 301_SWV3.0.4_C_F401, board V24S1_301, F4014 processor) the 2.1.2.2 bug. Cr-Touch crashes into the bed. Rolling back to 2.1.2.1 has resolved the issue

gitana2, 2024-02-29:
I confirm on Ender 3S1 (stock firmw 301_SWV3.0.4_C_F401, board V24S1_301, processor F401) the 2.1.2.2 bug. Cr-Touch crashes into the bed. Rolling back to 2.1.2.1 has resolved the issue

buff0k, 2024-02-29:
> I confirm on Ender 3S1 (stock firmw 301_SWV3.0.4_C_F401, board V24S1_301, processor F401) the 2.1.2.2 bug. Cr-Touch crashes into the bed. Rolling back to 2.1.2.1 has resolved the issue

Did you try the fix as per @ellensp above:

n 2.1.2.2 Marlin/src/inc/Conditionals_post.h is

 #if ALL(ENDSTOPPULLUPS, USE_Z_MIN_PROBE)
    #define ENDSTOPPULLUP_ZMIN_PROBE
  #endif
But USE_Z_MIN_PROBE is not part of 2.1.2.2, it came from bugfix by mistake.
So ENDSTOPPULLUP_ZMIN_PROBE is never defined.

Update this code to

  #if ALL(ENDSTOPPULLUPS, HAS_Z_MIN_PROBE_PIN)
    #define ENDSTOPPULLUP_ZMIN_PROBE
  #endif
And ENDSTOPPULLUP_ZMIN_PROBE gets defined as it should.

gitana2, 2024-03-07:
Thanks for the fix, but for now I will continue in 2121 without wasting time on new updates and tests. Maybe later...

bwskin, 2024-03-13:
I can confirm that with SKR2 this issue occurs on 2.1.2.2 and the patch provided by @ellensp fixes the issue. Thanks!

vfbank, 2024-03-16:
I'm using the MKS Tinybee board. This is an ESP32 based board.

Uses IR FIX sensor instead of BL touch. I don't think it's a BL Touch code issue. Because in 2.1.2.1 there is no such problem.

thisiskeithb, 2024-03-16:
> I don't think it's a BL Touch code issue. Because in 2.1.2.1 there is no such problem.

It's a general probe issue, not BLTouch-specific.

Phillip-0x01, 2024-03-17:
I was having the same issue on 2.1.2.2 with my Ender 3 V4.2.2 and BLTouch and the patch provided by @ellensp fixes the issue.

thisiskeithb, 2024-03-28:
Closing since this is already fixed in `bugfix-2.1.x`.

closed 2024-03-28 by thisiskeithb

phiphila, 2024-04-03:
> I think i've found something which looks like a very likely suspect to break many probes in the probe port.
> 
> In 2.1.2.2 Marlin/src/inc/Conditionals_post.h is
> 
> ```c++
>  #if ALL(ENDSTOPPULLUPS, USE_Z_MIN_PROBE)
>     #define ENDSTOPPULLUP_ZMIN_PROBE
>   #endif
> ```
> 
> But USE_Z_MIN_PROBE is not part of 2.1.2.2, it came from bugfix by mistake. So ENDSTOPPULLUP_ZMIN_PROBE is never defined.
> 
> Update this code to
> 
> ```c++
>   #if ALL(ENDSTOPPULLUPS, HAS_Z_MIN_PROBE_PIN)
>     #define ENDSTOPPULLUP_ZMIN_PROBE
>   #endif
> ```
> 
> And ENDSTOPPULLUP_ZMIN_PROBE gets defined as it should.
> 
> Can anyone with this issue very carefully test Z homing with this patch? @jradwan @buff0k

Hi 
There are truly great people on earth. Many thanks for sharing. this works for me.

My setup: printer : Ender5Plus with a mainboard creality 4.2.7 and a crtouch 5pins.
build with visual studio code.

[image]

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/14859294/Configuration.h.txt)

piotrmocko, 2024-04-19:
> I have an Antclabs BLTouch v3.1 connected to a BigTreeTech SKR Mini E3 v2. Can confirm with the above suggested patch the probe now works as expected.
> 
> ```diff
> diff --git a/Marlin/src/inc/Conditionals_post.h b/Marlin/src/inc/Conditionals_post.h
> index 6794087ca4..4e18fd3938 100644
> --- a/Marlin/src/inc/Conditionals_post.h
> +++ b/Marlin/src/inc/Conditionals_post.h
> @@ -2852,7 +2852,7 @@
>    #ifndef PROBE_OFFSET_YMAX
>      #define PROBE_OFFSET_YMAX  50
>    #endif
> -  #if ALL(ENDSTOPPULLUPS, USE_Z_MIN_PROBE)
> +  #if ALL(ENDSTOPPULLUPS, HAS_Z_MIN_PROBE_PIN)
>      #define ENDSTOPPULLUP_ZMIN_PROBE
>    #endif
>    #ifndef XY_PROBE_FEEDRATE
> ```

This has helped me to install 2.1.2.2 and not break my printer

reneroth, 2024-04-27:
Would it make sense to put a big warning re: this known bug on the release page? Took me quite a while to figure out what was going on 😨

thisiskeithb, 2024-04-28:
Patch for `2.1.x` submitted:
- https://github.com/MarlinFirmware/Marlin/issues/26809

TheGITofTeo997, 2024-04-28:
> Would it make sense to put a big warning re: this known bug on the release page? Took me quite a while to figure out what was going on 😨

Commenting here to say the same! I wasted a good amount of time and days trying to understand what was going on... until someone on Reddit suggested me this Git issue! All I did was downloading the latest release on the website and from there the bad things started. I definitely agree that this should be something advertised somewhere else. 
I compiled myself today the latest bugfix and luckily now all is working

thisiskeithb, 2024-04-28:
> I definitely agree that this should be something advertised somewhere else.

I created & pinned https://github.com/MarlinFirmware/Marlin/issues/27019 yesterday and just updated the `2.1.2.2` release notes to add a link to the pinned issue under the "Known Issues" heading:

[image]

TheGITofTeo997, 2024-04-28:
> > I definitely agree that this should be something advertised somewhere else.
> 
> I created & pinned #27019 yesterday and just updated the `2.1.2.2` release notes to add a link to the pinned issue under the "Known Issues" heading:
> <img alt="image" width="444" src="https://private-user-images.githubusercontent.com/13375512/326289854-cf95f4cf-5f79-46d0-96b6-70430aa9975e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQzMzQ4MDMsIm5iZiI6MTcxNDMzNDUwMywicGF0aCI6Ii8xMzM3NTUxMi8zMjYyODk4NTQtY2Y5NWY0Y2YtNWY3OS00NmQwLTk2YjYtNzA0MzBhYTk5NzVlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI4VDIwMDE0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYzYWE1NTlkM2M3OWIyODc2MWIzNjNhZGJhNjRlYmM4YjIxN2M0MDMzMzZiMzZjNTZmZTFjNmZhNjRhODI0NDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.YP7Q1Ttjxt9JQ31SMJTK0Wqdtdi0AohTlByGNFVtmIA">

I understand and thanks for this. What I am thinking about, is that maybe most of the people (sadly as I did) will simply download it from the Marlin website without being aware of this.
Sure it was my fault not checking the issues page, but I am afraid most of the people will do like I did.

reneroth, 2024-04-28:
Depending on POV, this might be a broken release and not a known issue 😬

HerbertExtrem, 2024-04-29:
> Depending on POV, this might be a broken release and not a known issue 😬

I installed my new 3D printer last week and this bug cost me a lot of time, since there are several reasons that could lead to nozzle crashing into bed. Since there is no workaround but changing the code i would clearly see this as a broken release.

Weirdo50, 2024-05-02:
> I created & pinned #27019 yesterday and just updated the `2.1.2.2` release notes to add a link to the pinned issue under the "Known Issues" heading:
>
> <img alt="image" width="444" [...]

auto build marlin does not detect the new version.

gitana2, 2024-05-06:
Sorry for the delay. I confirm that @ellensp's fix in 2.1.2.2 Marlin/src/inc/Conditionals_post.h which replaces "#if ALL(ENDSTOPPULLUPS, USE_Z_MIN_PROBE)" with #if ALL(ENDSTOPPULLUPS, HAS_Z_MIN_PROBE_PIN) fixes the problem on Ender 3S1, board V24S1_301 , processor F401. Congratulations!

SojournTV, 2024-05-29:
Thanks for the fix!

HerbertExtrem, 2024-05-29:
Am I missing something here or this Fix is not Part of any Branch / Release? Or did this get fixed with a different ticket? Can't find any info in which Commits on which Branches this is included. Doesn't seem to be contained in 2.1.2.3? Thx and greetings

damianshaon, 2024-05-30:
Fix worked for me. many thanks......

ellensp, 2024-05-30:
This patch and others are included in Marlin 2.1.2.3 https://github.com/MarlinFirmware/Marlin/archive/2.1.2.3.zip

HerbertExtrem, 2024-05-30:
> This patch and others are included in Marlin 2.1.2.3 https://github.com/MarlinFirmware/Marlin/archive/2.1.2.3.zip

Thanks, already got it then. Just still don't understand where or if I can see a reference to #27022 from this ticket :)

thisiskeithb, 2024-05-30:
> Thanks, already got it then. Just still don't understand where or if I can see a reference to #27022 from this ticket :)

Scroll up:

[image]