MarlinFirmware/Marlin #26817: [BUG] #define Y2_DRIVER_TYPE does not work in 2.1.2.2

vfbank, 2024-02-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

#define HAS_DUAL_Y_STEPPERS 1 in Marlin/src/inc/Conditionals_LCD.h, which existed until 2.1.2.1, is missing.

I still need dual mode. Is there any reason to delete this existing code?

### Version of Marlin Firmware

2.1.2.2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-02-25:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/26801

closed 2024-02-25 by ellensp

vfbank, 2024-02-25:
> 중복 #26801

What's wrong with #define HAS_DUAL_Y_STEPPERS 1? What I'm curious about is why this syntax was removed in version 2.1.2.2.