MarlinFirmware/Marlin #26830: > Compilation fails. Using the old Octopus Pro 1.0 definitions results in an STM32F4 error, and using the Octopus Pro Max EZ definitions compiles but crashes the firmware.

FREAKASAS, 2024-03-02:
[image]
[image]
              > Compilation fails. Using the old Octopus Pro 1.0 definitions results in an STM32F4 error, and using the Octopus Pro Max EZ definitions compiles but crashes the firmware.

You don't reuse any of the Octopus Pro 1.0 / STM32F4 definitions with the referenced PR.

Use the files from [the PR's source fork](https://github.com/thisiskeithb/Marlin/tree/pr/add_octopus_pro_1_1_stm32H723zet6) ([.ZIP download](https://github.com/thisiskeithb/Marlin/archive/refs/heads/pr/add_octopus_pro_1_1_stm32H723zet6.zip)) or wait until it is merged.

Using a minimal config, this compiles fine.

<details><summary>Minimal Config:</summary>
<p>

```diff
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index b21eba7..2c7199f 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -67,7 +67,7 @@
 
 // Choose the name from boards.h that matches your setup
 #ifndef MOTHERBOARD
-  #define MOTHERBOARD BOARD_RAMPS_14_EFB
+  #define MOTHERBOARD BOARD_BTT_OCTOPUS_PRO_V1_1
 #endif
 
 /**
@@ -78,7 +78,7 @@
  *
  * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
  */
-#define SERIAL_PORT 0
+#define SERIAL_PORT -1
 
 /**
  * Serial Port Baud Rate
diff --git a/platformio.ini b/platformio.ini
index e3bdb6f..f54fee7 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -13,7 +13,7 @@
 [platformio]
 src_dir      = Marlin
 boards_dir   = buildroot/share/PlatformIO/boards
-default_envs = mega2560
+default_envs = STM32H723ZE_btt
 include_dir  = Marlin
 extra_configs =
     Marlin/config.ini
```

</p>
</details>

Compile log using the above minimal config:

<details><summary>STM32H723ZE_btt  SUCCESS:</summary>
<p>

```prolog
Archiving .pio/build/STM32H723ZE_btt/libFrameworkArduino.a
Indexing .pio/build/STM32H723ZE_btt/libFrameworkArduino.a
Linking .pio/build/STM32H723ZE_btt/firmware.elf
Checking size .pio/build/STM32H723ZE_btt/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   2.0% (used 11544 bytes from 577536 bytes)
Flash: [==        ]  20.0% (used 104916 bytes from 524288 bytes)
Building .pio/build/STM32H723ZE_btt/firmware.bin
====================================== [SUCCESS] Took 40.74 seconds ======================================

Environment      Status    Duration
---------------  --------  ------------
STM32H723ZE_btt  SUCCESS   00:00:40.735
======================================= 1 succeeded in 00:00:40.735 =======================================
 *  Terminal will be reused by tasks, press any key to close it. 
```

</p>
</details>

_Originally posted by @thisiskeithb in https://github.com/MarlinFirmware/Marlin/issues/26044#issuecomment-1618362922_

ellensp, 2024-03-02:
@FREAKASAS Please check your issue, most links do not work and it is rather incomprehensible

Also please use the issue template when creating issues... 

Desperately trying to read between the lines...

Building with #define MOTHERBOARD BOARD_BTT_OCTOPUS_PRO_V1_0

This has 5 different build environments.

✅ STM32F446ZE_btt
✅ STM32F446ZE_btt_usb_flash_drive
✅STM32F429ZG_btt
✅STM32F429ZG_btt_usb_flash_drive
❌STM32H723ZE_btt

4 build fine but STM32H723ZE_btt results in #error "Oops! Select an STM32F4 board in 'Tools > Board.'"

Marlin/src/pins/stm32f4/pins_BTT_OCTOPUS_V1_common.h needs updated 

```CPP
#include "env_validate.h"
```
becomes 

```CPP
#if NOT_TARGET(STM32H7)
  #include "env_validate.h"
#else
  #include "../stm32h7/env_validate.h"
#endif
```
To allow for the STM32H7 Variant.

I cannot test any STM32H723 based board as I don't have access to any.

thisiskeithb, 2024-03-02:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/26831

closed 2024-03-02 by thisiskeithb