MarlinFirmware/Marlin #26833: [BUG] Compilation of WF with 2 extruders.

radek8, 2024-03-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have a printer with 2 nozzles. I regularly update to new versions.
Now the compilation ends with an error.
I found that it brought the error
https://github.com/MarlinFirmware/Marlin/pull/26152

### Bug Timeline

24.1.2024

### Expected behavior

Compilation of FW

### Actual behavior

Compilation error

[image]

After commenting this line 
u8g.drawBitmapP(_MAX(0, STATUS_HOTEND_X(heater_id) - 6), STATUS_HEATERS_Y + 3, 1, 5, status_active_extruder_indicator_bmp);
in the status_screen_DOGM.cpp file, the compilation will succeed

### Version of Marlin Firmware

bugfix 2.1.x

### Printer model

Ender3 - modified

### Electronics

SKR V1.4 Turbo

### LCD/Controller

BTT TFT 3.5 E3 V3.0

### Other add-ons

Print head - pendulum

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/14470594/config.zip)

thisiskeithb, 2024-03-02:
Your config does not compile with stock Marlin, so please attach your modified pins file / any other modified files:

```prolog
Marlin/src/HAL/shared/../../inc/SanityCheck.h:714:6: error: #error "SERVO1_PIN must be defined for your SWITCHING_NOZZLE."
  714 |     #error "SERVO1_PIN must be defined for your SWITCHING_NOZZLE."
      |      ^~~~~
Marlin/src/HAL/shared/../../inc/SanityCheck.h:756:6: error: #error "SERVO1_PIN must be defined for your SWITCHING_EXTRUDER."
  756 |     #error "SERVO1_PIN must be defined for your SWITCHING_EXTRUDER."
      |      ^~~~~
Marlin/src/HAL/LPC1768/../../inc/SanityCheck.h:964:26: error: static assertion failed: NUM_SERVOS (or some servo index) is too large. The selected board only has 1 servos.
```

radek8, 2024-03-02:
[Pins.zip](https://github.com/MarlinFirmware/Marlin/files/14470612/Pins.zip)

thisiskeithb, 2024-03-02:
I had to disable `SHOW_CUSTOM_BOOTSCREEN`, but I can now reproduce the error message.

Disable `STATUS_HOTEND_NUMBERLESS` and it'll compile.

radek8, 2024-03-02:
It works.
I really wouldn't have found this on my own.
Thank you for your time.
Maybe the bug could point this out so that other users are also alerted.

thisiskeithb, 2024-03-02:
Alternatively, you can leave `STATUS_HOTEND_NUMBERLESS` enabled and patch `status_screen_DOGM.cpp`:

```diff
diff --git a/Marlin/src/lcd/dogm/status_screen_DOGM.cpp b/Marlin/src/lcd/dogm/status_screen_DOGM.cpp
index 2416501..e77a0f1 100644
--- a/Marlin/src/lcd/dogm/status_screen_DOGM.cpp
+++ b/Marlin/src/lcd/dogm/status_screen_DOGM.cpp
@@ -323,7 +323,7 @@ FORCE_INLINE void _draw_centered_temp(const celsius_t temp, const uint8_t tx, co
 
       } // PAGE_CONTAINS
 
-      #if HAS_MULTI_EXTRUDER && DISABLED(SLIM_LCD_MENUS)
+      #if HAS_MULTI_EXTRUDER && NONE(SLIM_LCD_MENUS, STATUS_HOTEND_NUMBERLESS)
         if (active_extruder == heater_id)
           u8g.drawBitmapP(_MAX(0, STATUS_HOTEND_X(heater_id) - 6), STATUS_HEATERS_Y + 3, 1, 5, status_active_extruder_indicator_bmp);
       #endif
```

I'll submit a PR with this patch shortly.

thisiskeithb, 2024-03-02:
PR submitted, so please give https://github.com/MarlinFirmware/Marlin/pull/26834 a try.

closed 2024-03-02 by thisiskeithb