MarlinFirmware/Marlin #26836: [BUG] Homing Fails on latest bugfix 2.1.x nightly, if FT Motion enabled.

jsondag, 2024-03-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

If Fixed-Time Motion is enabled, printer will move up, then home X, and stop with endstop engaged without backing off, then will not home Y, and not move to the center to probe Z, but probe Z from the location it engaged the X endstop.

### Bug Timeline

Started when first enabled FT Motion

### Expected behavior

Move up, move to X endstop, back off from Endstop, then move to Y endstop, back off from Y, then move Probe to center of bed, and probe bed.

### Actual behavior

Move up, then home X, and stop with endstop engaged without backing off, then will not home Y, and not move to the center to probe Z, but probe Z from the location it engaged the X endstop.

### Steps to Reproduce

M493 S17
G28

### Version of Marlin Firmware

Marlin bugfix-2.1.x (Mar  2 2024 18:32:15) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:HEVO 500 KINEMATICS:COREXYCartesian EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Printer model

Hypercube Evolution custom

### Electronics

BTT SKR 1.4 Turbo

### LCD/Controller

BTT TFT70

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/14471231/Configuration.zip)

thisiskeithb, 2024-03-03:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/26683

thisiskeithb, 2024-03-03:
This will be fixed in https://github.com/MarlinFirmware/Marlin/pull/26720

closed 2024-03-03 by thisiskeithb