MarlinFirmware/Marlin #26838: [BUG] Reset when upload file with Cura MKS WIFI plugin

chrbcd, 2024-03-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When use Cura for Slice and upload file, The printer reset. Error with V2.2.2.2 and  V bugfix-2.1x.

### Bug Timeline

new bug since V2.1.2.1

### Expected behavior

The file must is uploaded and print.

### Actual behavior

At the time of transfer, the printer screen displays "Uploading" then triggers a reset. Error message on the PC.

### Steps to Reproduce

Turn on the printer. 
No manipulation on the printer. 
With CURA, open an STL file, slicet it then request transfer via Wifi

### Version of Marlin Firmware

bugfix-2.1x (loaded 03/02/2024) and 2.1.2.2

### Printer model

Wanaho D12-300

### Electronics

MKS_ROBIN_NANO_V3_1

### LCD/Controller

TS35

### Other add-ons

none

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Attached is the file /Marlin/src/lcd/extui/mks_ui/wifi_module.cpp with the corrections I made. (Ok now)
[config and source wifi_module modified.zip](https://github.com/MarlinFirmware/Marlin/files/14472677/config.and.source.wifi_module.modified.zip)

ellensp, 2024-03-04:
Can you try latest marlin,  changes where made 15 hours ago which may change things

chrbcd, 2024-03-04:
I just tested, still the same problem. For the Cura plugin which only manages the SD card, only 2 lines need to be modified in WiFiModule.cpp:

720  if (gCfgItems.fileSysType == FILE_SYS_SD) {
721 +  card.changeMedia(&card.media_driver_sdcard); 
722    TERN_(HAS_MEDIA, card.mount());
723  }

And to have the list of files:

840 - strcat_P(Fstream, PSTR(" 0")); // report 0 file size

 The other modifications that I made in the file transmitted yesterday only concern the use of the USB port and the management of the exploring subdirectories.
Thanks

closed 2024-06-17 by github-actions[bot]