MarlinFirmware/Marlin #26840: [BUG] (Delta X/Y movement)

renzaaa, 2024-03-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Z movement works fine, but any X or Y movement moves all 3 axes equally down until hits the bed.
The same problem as [BUG] Delta X/Y movement broken #26383.

### Version of Marlin Firmware

2.1.2.2

### Printer model

Custom Delta

### Electronics

SKR_V1_4_TURBO

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/14475344/Config.zip)

ellensp, 2024-03-04:
May be related https://github.com/MarlinFirmware/Marlin/issues/26792

thisiskeithb, 2024-06-02:
This should now be fixed in the latest [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip), so please test with the latest code and let us know if you're still having this issue.

lenny-chem-dat, 2024-06-25:
> This should now be fixed in the latest [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip), so please test with the latest code and let us know if you're still having this issue.

I can confirm that the Delta movement works fine on the version I downloaded a week ago. It didn't work before, so bug seems to be fixed. Running an FLSUN Q5 with MKS Robin Nano.

closed 2024-06-25 by thisiskeithb