MarlinFirmware/Marlin #26843: [BUG] STM32F4 "'spi_mode_e' does not name a type"

WillShawx, 2024-03-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When trying to compile a new build to an BTT Octopus Pro, getting error "'spi_mode_e' does not name a type"

Attempted bug fix from reddit - replace buildroot/share/PlatformIO/scripts/generic_create_variant.py with the one in bugfix https://raw.githubusercontent.com/MarlinFirmware/Marlin/bugfix-2.1.x/buildroot/share/PlatformIO/scripts/generic_create_variant.py

### Bug Timeline

New Error

### Expected behavior

I Expected It To Compile

### Actual behavior

Error "xx" presented. 

### Steps to Reproduce

1. Download attached ZIP file
2. Attempt compilation

### Version of Marlin Firmware

2.1.2.2 

### Electronics

BTT Octopus Pro v1.1

### LCD/Controller

BTT TFT35 E3

### Bed Leveling

ABL 3-point

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin-2.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/14495029/Marlin-2.1.x.zip)

WillShawx, 2024-03-05:
In file included from /Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/MarlinSPI.cpp:27:
/Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/MarlinSPI.h:99:3: error: 'spi_mode_e' does not name a type
   99 |   spi_mode_e _dataMode;
      |   ^~~~~~~~~~
/Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/MarlinSPI.h: In constructor 'MarlinSPI::MarlinSPI(pin_t, pin_t, pin_t, pin_t)':
/Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/MarlinSPI.h:64:5: error: '_dataMode' was not declared in this scope; did you mean '_dataSize'?
   64 |     _dataMode = SPI_MODE_0;
      |     ^~~~~~~~~
      |     _dataSize
/Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/MarlinSPI.h:64:17: error: 'SPI_MODE_0' was not declared in this scope; did you mean 'SPI_MODE0'?
   64 |     _dataMode = SPI_MODE_0;
      |                 ^~~~~~~~~~
      |                 SPI_MODE0
/Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/MarlinSPI.h: In member function 'void MarlinSPI::setDataMode(uint8_t)':
/Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/MarlinSPI.h:83:23: error: '_dataMode' was not declared in this scope; did you mean '_dataSize'?
   83 |       case SPI_MODE0: _dataMode = SPI_MODE_0; break;
      |                       ^~~~~~~~~
      |                       _dataSize
/Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/MarlinSPI.h:83:35: error: 'SPI_MODE_0' was not declared in this scope; did you mean 'SPI_MODE0'?
   83 |       case SPI_MODE0: _dataMode = SPI_MODE_0; break;
      |                                   ^~~~~~~~~~
      |                                   SPI_MODE0
/Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/MarlinSPI.h:84:35: error: 'SPI_MODE_1' was not declared in this scope; did you mean 'SPI_MODE1'?
   84 |       case SPI_MODE1: _dataMode = SPI_MODE_1; break;
      |                                   ^~~~~~~~~~
      |                                   SPI_MODE1
/Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/MarlinSPI.h:85:35: error: 'SPI_MODE_2' was not declared in this scope; did you mean 'SPI_MODE2'?
   85 |       case SPI_MODE2: _dataMode = SPI_MODE_2; break;
      |                                   ^~~~~~~~~~
      |                                   SPI_MODE2
/Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/MarlinSPI.h:86:35: error: 'SPI_MODE_3' was not declared in this scope; did you mean 'SPI_MODE3'?
   86 |       case SPI_MODE3: _dataMode = SPI_MODE_3; break;
      |                                   ^~~~~~~~~~
      |                                   SPI_MODE3
/Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/MarlinSPI.cpp: At global scope:
/Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/MarlinSPI.cpp:29:50: error: 'spi_mode_e' has not been declared
   29 | static void spi_init(spi_t *obj, uint32_t speed, spi_mode_e mode, uint8_t msb, uint32_t dataSize) {
      |                                                  ^~~~~~~~~~
/Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/MarlinSPI.cpp: In function 'void spi_init(spi_t*, uint32_t, int, uint8_t, uint32_t)':
/Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/MarlinSPI.cpp:30:24: error: invalid conversion from 'int' to 'SPIMode' [-fpermissive]
   30 |   spi_init(obj, speed, mode, msb);
      |                        ^~~~
      |                        |
      |                        int
In file included from /Users/willshawcross/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.7.1/libraries/SPI/src/SPI.h:18,
                 from /Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/MarlinSPI.h:25:
/Users/willshawcross/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.7.1/libraries/SPI/src/utility/spi_com.h:85:51: note:   initializing argument 3 of 'void spi_init(spi_t*, uint32_t, SPIMode, uint8_t)'
   85 | void spi_init(spi_t *obj, uint32_t speed, SPIMode mode, uint8_t msb);
      |                                           ~~~~~~~~^~~~
/Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/MarlinSPI.cpp: In member function 'void MarlinSPI::begin()':
/Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/MarlinSPI.cpp:47:27: error: '_dataMode' was not declared in this scope; did you mean '_dataSize'?
   47 |   spi_init(&_spi, _speed, _dataMode, _bitOrder, _dataSize);
      |                           ^~~~~~~~~
      |                           _dataSize
/Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/timers.cpp:328:40: error: static assertion failed: One or more timer conflict detected. Examine "timers_in_use" to help identify conflict.
  328 | static_assert(verify_no_timer_conflicts(), "One or more timer conflict detected. Examine \"timers_in_use\" to help identify conflict.");
      |               ~~~~~~~~~~~~~~~~~~~~~~~~~^~
In file included from /Users/willshawcross/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.7.1/cores/arduino/stm32/clock.h:19,
[... cut ...]
/Users/willshawcross/Documents/3D Printing/Marlin-bugfix-2.1.x/Marlin/src/HAL/STM32/HAL.cpp:159:14: note: in expansion of macro 'LED_PIN'
  159 |       TOGGLE(LED_PIN);  // heartbeat indicator
      |              ^~~~~~~
Multiple libraries were found for "Servo.h"
  Used: /Users/willshawcross/Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.7.1/libraries/Servo
  Not used: /Users/willshawcross/Library/Arduino15/libraries/Servo
exit status 1

Compilation error: 'spi_mode_e' does not name a type

ellensp, 2024-03-05:
Provided code is just a mess

"#define LED_PIN                             PBO"   (should be 0 not a capital o) 

assorted updates to Marlin/src/HAL/STM32/MarlinSPI.h that are not even valid code

Listed "fix" you tried is unrelated

The board BTT Octopus Pro v1.1 is not in Marlin 2.1.2.2

BOARD_BTT_OCTOPUS_PRO_V1_1 is only available in bugfix 2.1.x at this time

 
Error spi_mode_e is due to Arduino_Core_STM32 using latest version not the version that is compatible with marlin

needs 2.6.0 not 2.7.0

see https://github.com/stm32duino/Arduino_Core_STM32/compare/2.6.0...2.7.0 

But that error is the least of your issues....

ellensp, 2024-03-05:
update ini/stm32h7.ini to force platformio to use 2.6.0 

[env:STM32H723ZE_btt]
extends                     = STM32H723Zx_btt
board                       = marlin_STM32H723ZE
platform_packages           = framework-arduinoststm32@https://github.com/stm32duino/Arduino_Core_STM32/archive/refs/tags/2.6.0.zip

closed 2024-03-05 by thisiskeithb