MarlinFirmware/Marlin #26847: [BUG] TFT_LVGL_UI + BROWSE_MEDIA_ON_INSERT = compilation errors with no human-readable text

l-e-s, 2024-03-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I bought a new mainboard for my printer and compiling firmware for it:
```
#define SDSUPPORT
#define TFT_LVGL_UI
#if ENABLED(SDSUPPORT)
  ...
  #define BROWSE_MEDIA_ON_INSERT
```
This combination of flags results in a compilation errors. Compilation succeeded if i change TFT_LVGL_UI to TFT_COLOR_UI or disable BROWSE_MEDIA_ON_INSERT. ("Marlin/src/lcd/marlinui.cpp:1781:11: error: 'clear_menu_history' was not declared in this scope")

### Bug Timeline

Checked lts-2.1.0, bugfix-2.1.x with the same result

### Expected behavior

TFT_LVGL_UI compiled with the BROWSE_MEDIA_ON_INSERT support

### Actual behavior

```
...
Compiling .pio/build/mks_robin_nano_v3_1/src/src/lcd/tft_io/touch_calibration.cpp.o
Compiling .pio/build/mks_robin_nano_v3_1/src/src/lcd/utf8.cpp.o
Compiling .pio/build/mks_robin_nano_v3_1/src/src/libs/W25Qxx.cpp.o
Marlin/src/lcd/extui/mks_ui/wifi_module.cpp: In function 'uint8_t exploreDisk(const char*, uint8_t, bool)':
Marlin/src/lcd/extui/mks_ui/wifi_module.cpp:829:14: warning: unused variable 'diveDir' [-Wunused-variable]
  829 |   MediaFile *diveDir;
      |              ^~~~~~~
Compiling .pio/build/mks_robin_nano_v3_1/src/src/libs/buzzer.cpp.o
Compiling .pio/build/mks_robin_nano_v3_1/src/src/libs/crc16.cpp.o
Compiling .pio/build/mks_robin_nano_v3_1/src/src/libs/nozzle.cpp.o
Marlin/src/lcd/marlinui.cpp: In static member function 'static void MarlinUI::media_changed(uint8_t, uint8_t)':
Marlin/src/lcd/marlinui.cpp:1781:11: error: 'clear_menu_history' was not declared in this scope
 1781 |           clear_menu_history();
      |           ^~~~~~~~~~~~~~~~~~
Compiling .pio/build/mks_robin_nano_v3_1/src/src/libs/numtostr.cpp.o
In file included from Marlin/src/lcd/../inc/MarlinConfigPre.h:37,
                 from Marlin/src/lcd/../inc/MarlinConfig.h:28,
                 from Marlin/src/lcd/marlinui.cpp:23:
Marlin/src/lcd/marlinui.h:256:91: error: 'menu_media' was not declared in this scope
  256 |     #define MEDIA_MENU_GATEWAY TERN(PASSWORD_ON_SD_PRINT_MENU, password.media_gatekeeper, menu_media)
      |                                                                                           ^~~~~~~~~~
Marlin/src/lcd/../inc/../core/macros.h:672:26: note: in definition of macro 'THIRD'
  672 | #define THIRD(a,b,c,...) c
      |                          ^
Marlin/src/lcd/../inc/../core/macros.h:251:29: note: in expansion of macro '___TERN'
  251 | #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
      |                             ^~~~~~~
Marlin/src/lcd/../inc/../core/macros.h:250:29: note: in expansion of macro '__TERN'
  250 | #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
      |                             ^~~~~~
Marlin/src/lcd/../inc/../core/macros.h:246:29: note: in expansion of macro '_TERN'
  246 | #define TERN(O,A,B)         _TERN(_ENA_1(O),B,A)    // OPTION ? 'A' : 'B'
      |                             ^~~~~
Marlin/src/lcd/marlinui.h:256:32: note: in expansion of macro 'TERN'
  256 |     #define MEDIA_MENU_GATEWAY TERN(PASSWORD_ON_SD_PRINT_MENU, password.media_gatekeeper, menu_media)
      |                                ^~~~
Marlin/src/lcd/marlinui.cpp:1783:23: note: in expansion of macro 'MEDIA_MENU_GATEWAY'
 1783 |           goto_screen(MEDIA_MENU_GATEWAY);
      |                       ^~~~~~~~~~~~~~~~~~
Marlin/src/lcd/marlinui.cpp:1783:11: error: 'goto_screen' was not declared in this scope
 1783 |           goto_screen(MEDIA_MENU_GATEWAY);
      |           ^~~~~~~~~~~
Compiling .pio/build/mks_robin_nano_v3_1/src/src/libs/stopwatch.cpp.o
*** [.pio/build/mks_robin_nano_v3_1/src/src/lcd/marlinui.cpp.o] Error 1
```

### Version of Marlin Firmware

2.1.2.2

### Printer model

Self-made from Anet A6

### Electronics

MKS Robin Nano 3.1

### LCD/Controller

MKS TS35 v2.0

### Bed Leveling

None

### Your Slicer

None

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

l-e-s, 2024-03-06:
[MKS_Robin_Nano.tar.gz](https://github.com/MarlinFirmware/Marlin/files/14512606/MKS_Robin_Nano.tar.gz)

thisiskeithb, 2024-03-06:
`BROWSE_MEDIA_ON_INSERT` is not supported on `TFT_LVGL_UI`, so you'll need to disable `BROWSE_MEDIA_ON_INSERT` or use another UI.