MarlinFirmware/Marlin #26850: [BUG] Skew correction is not applying at all.

Fercik86, 2024-03-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have been trying to get skew correction working on Ender 3. Initially I was running 2.0.9.3 firmware where M852 was showing as M851 I though that might be the issue. After upgrading to latest version skew is still not applying. I am trying to implement it mainly with:
#define XY_SKEW_FACTOR
so that it can be changed later.
I have no idea what might be causing it to not apply at all.

### Bug Timeline

For a long time. In version 2.0.9.3 as well.

### Expected behavior

When skew correction is set with M852 I expect some change in the printed part dimensions.

### Actual behavior

Skew result is always the same from the test print even with vastly different skew factors.

### Steps to Reproduce

Just printing.

### Version of Marlin Firmware

2.1.2.2

### Printer model

Ender-3 pro

### Electronics

BTT SKR V1.3

### Other add-ons

double Z

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/14528380/configs.zip)

Coizado, 2024-03-12:
It's working for me, at least the XY, I don't use the ones involving Z.

One must keep in mind that XY_SKEW_FACTOR overrides the measurements, and M852 overrides XY_SKEW_FACTOR, if you never set a value for M852 it will revert to XY_SKEW_FACTOR, but if you set M852 to 0, for example, and save to EEPROM, no matter how much you change XY_SKEW_FACTOR and reflash, it will be set to 0 at startup when the EEPROM values are loaded. EEPROM must be cleared for the value to revert back to XY_SKEW_FACTOR.

I'm on an Ender-3 v1, SKR mini E3 v1.2, latest bugfix, BLtouch clone, Octoprint.

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/14575124/configs.zip)

thisiskeithb, 2024-06-08:
> It's working for me, at least the XY, I don't use the ones involving Z.

It's been a while since I've ran this on my Ender-3 since I ended up fixing the leaning Z gantry issue at a hardware level, but I'd say that the skew correction still works.

Using our Simulator with a basic 25mm cube stretched to max Z height, skew factors work as intended:

### XY skew correction with a factor of 0.5 (`M852 I0.5`):

[image]

### XZ skew correction with a factor of 0.5 (`M852 J0.5`):

[image]

### YZ skew correction with a factor of 1.0 (`M852 K1.0`):

[image]

### All skew correction disabled (`M852 I0.0 J0.0 K0.0`):

[image]

closed 2024-06-08 by thisiskeithb