MarlinFirmware/Marlin #26851: [BUG] Use Junction Deviation - what happens to non-XY Jerk settings?

Sophist-UK, 2024-03-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

In v2.1.2.2 when I comment out the `#define CLASSIC_JERK` to enable Junction Deviation, the entire `Configuration/Advanced Settings/Jerk` menu is replaced with a single `Configuration/Advanced Settings/Junction Dev` setting.

I am confused now because:

1. I was of the belief that Junction Deviation replaces only X and Y jerk, and thus Z and E jerk (++) should remain valid. Is this the case or not? (In `Configuration.h` `DEFAULT_EJERK` still gets defined but `DEFAULT_ZJERK` doesn't get defined.)
2. If this is the case, how is ZJERK defined in your configuration?

- Don't we still need a menu so we can adjust Z and E Jerks (++) i.e. the Classic Jerk menu less X & Y Jerks plus Junction Deviation radius?,

### Bug Timeline

No idea

### Expected behavior

1. All `DEFAULT_*JERK`  are defined except X/Y regardless of the `CLASSIC_JERK` setting, with only `DEFAULT_XJERK` and `DEFAULT_YJERK` conditional on `CLASSIC_JERK`.

2. If `DISABLED(CLASSIC_JERK)` then `Configuration/Advanced Settings/Jerk` menu is retained minus `X Jerk` and `Y Jerk` and plus `Junction Dev.`.

### Actual behavior

Jerk menu is completely replaced.

### Steps to Reproduce

Comment out the `#define CLASSIC_JERK` statement.

### Version of Marlin Firmware

2.1.2.2

### Printer model

Dagoma DE200

### Electronics

MKS BASE 1.5

### LCD/Controller

REPRAPDISCOUNT_FULL_GRAPHICS

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

EvilGremlin, 2024-03-07:
This is not a bug. JD replace jerk completely. EJERK also not used anywhere in the code anymore with JD, actual bug would be EJERK still appearing in the menu.

thisiskeithb, 2024-03-07:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2024-03-07 by thisiskeithb

Sophist-UK, 2024-03-07:
Ok - so can someone explain or point me to documentation that explains what ZJERK and EJERK are replaced by?

Linear Advance has a requirement that E_JERK is >= 10 so how is this impacted?

I am very confused...

thisiskeithb, 2024-03-07:
> Linear Advance has a requirement that E_JERK is >= 10

You’re not running the latest [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) because that is no longer the case. Even in older code, you could override the sanity check by enabling `ALLOW_LOW_EJERK`.

Join our Discord if you have additional questions since we prefer not to handle user-support questions here.