MarlinFirmware/Marlin #26852: [BUG] (bug summary)Auto Align not compensate

demetan71, 2024-03-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When I was using endstop switch for homing Z auto align work, but now I'm homing Z with probe, G34 do it's move but do not compensate adjust...

### Bug Timeline

Since I' homing Z with probe

### Version of Marlin Firmware

2.1.x

### Printer model

Anycubic Chiron

### Electronics

SKR 1.4 Turbo 

### LCD/Controller

TFT35 E3 V3

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I'm desesperate, could'nt auto align Z so my bed leveling not good or I couldn't verify it.....

Try almost everything I made a video to show what it does

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/14530278/Configuration_adv.zip)

https://www.youtube.com/watch?v=c4JiuWvfRgY

thisiskeithb, 2024-03-07:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as [Pronterface](https://github.com/kliment/Printrun/releases/latest), [Repetier Host](https://www.repetier.com/download-now/), or [OctoPrint](https://octoprint.org/download/).
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

demetan71, 2024-03-07:
[thisiskeithb](https://github.com/thisiskeithb)

Ok, but Z auto align Will give me a report???   Because bed leveling seem's to work ok, but so much difference between to end of the axis, so it's hard to validate that bed leveling is ok....

thisiskeithb, 2024-03-07:
> Ok, but Z auto align Will give me a report??? Because bed leveling seem's to work ok, but so much difference between to end of the axis, so it's hard to validate that bed leveling is ok....

I've updated my comment. Add `G34` after `G29` so it can be logged.

demetan71, 2024-03-08:
[thisiskeithb](https://github.com/thisiskeithb)

Ok I downlaoded the last bugfix version
Is it normal that when I send G29 by pronterface it does nothing? I have to do G29 P1

Did all what you said, but I want to copy the report, I scroll up, and I can't find the beginnng, I just have report in the middle of G34???

What I did wrong??

demetan71, 2024-03-08:
https://github.com/thisiskeithb

Finally I switch the wire motor on the board, and I can acheive accuracy of 0.01 with 30 itteration!!

So now it's working, but how why it take 30 itterations to do it??   Starting from .322 to 0.01

closed 2024-03-08 by thisiskeithb

demetan71, 2024-03-09:
https://github.com/thisiskeithb

I'm feeling a little shy, shame on me, the problem was that my second Z stepper was connected on Z2 instead of E1 so now it's working.  I was able to achieve perfect accuracy 0.00, but it's not perfect somethime I have to do it a couple of time, because sometime it compensate in the wrong direction an I've got the message Accuracy decreasing....

But that doesn't resoleve my main issue it to print on a perfect leveling bed

Since I put the last bug fix firmware I've got message Probing fail, when I put Bi-Linear it simply said unable to enable bed leveling, so I switch to Linear and I have probing fail...  I have report but unble to interpret it...
[probe-fail.txt](https://github.com/MarlinFirmware/Marlin/files/14547860/probe-fail.txt)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/14547864/Configuration_adv.zip)