MarlinFirmware/Marlin #26858: [BUG] Enabling FT_MOTION with any shaper causes grinding, ignores endstops.

bmourit, 2024-03-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Creality board 4.2.2 with A4988 drivers enabling FT_MOTION will cause movements to attempt moving beyond the endstops and motor grinding will occure.  Hardware endstops endabled, but ignored. Doesn't seem to matter which shaper is used, problem persists.

### Expected behavior

No grinding

### Actual behavior

Grinding/movement beyond endstops

### Version of Marlin Firmware

Latest Bugfix branch

### Printer model

Ender-3 V4.2.2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/14549810/Archive.zip)

narno2202, 2024-03-10:
Your config files are not from bugfix but  a mix from PR #26848 and from other branch.  What is your Marlin version?
Please report here only bugs from bugfix  or for Marlin stable.

narno2202, 2024-03-10:
@bmourit , please post your config files for bugfix. It's look like a buffer issue ( buffers too large).

bmourit, 2024-03-14:
> @bmourit , please post your config files for bugfix. It's look like a buffer issue ( buffers too large).

I believe you are correct! Feel free to close this. I'll open a new bug if I find it again, but this seems to have been the issue. Thanks so much!

closed 2024-03-14 by bmourit