MarlinFirmware/Marlin #26859: [BUG] DGUS_LCD_UI_RELOADED not compiling Marlin 2.1.2.2

Ale11Re, 2024-03-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When selecting DGUS_LCD_UI_RELOADED to compile with CR10s Pro. I'm providing my configurations. It does compile with Marlin 2.1.2.1. I did also try with latest Marlin bugfix 2.1.x but does not compile due to other errors. Compilation error: ExtUI not defined. When commenting line 111 of /src/feature/powerloss.cpp "TERN_(EXTENSIBLE_UI, ExtUI::onSetPowerLoss(enabled));" It compiles fine.
[image]

### Expected behavior

Expect Marlin 2.1.2.2 to compile

### Actual behavior

Marlin 2.1.2.2 won't compile

### Steps to Reproduce

With

### Version of Marlin Firmware

2.1.2.2

### Printer model

Creality CR10S Pro

### LCD/Controller

DGUS_LCD_UI_RELOADED

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Conf.zip](https://github.com/MarlinFirmware/Marlin/files/14551452/Conf.zip)

ellensp, 2024-03-10:
The ExtUI headers are not included...

apply this diff

```DIFF
diff --git a/Marlin/src/feature/powerloss.cpp b/Marlin/src/feature/powerloss.cpp
index 86e6b780bd..69e23f4a32 100644
--- a/Marlin/src/feature/powerloss.cpp
+++ b/Marlin/src/feature/powerloss.cpp
@@ -31,6 +31,10 @@
 #include "powerloss.h"
 #include "../core/macros.h"
 
+#if ENABLED(EXTENSIBLE_UI)
+  #include "../lcd/extui/ui_api.h"
+#endif
+
 bool PrintJobRecovery::enabled; // Initialized by settings.load()
 
 MediaFile PrintJobRecovery::file;

```

ellensp, 2024-03-10:
Issue is not present in bugfix

Ale11Re, 2024-03-11:
My bad, if not present in bugfix I screwed up something else

thisiskeithb, 2024-03-28:
Closing since this is already fixed in `bugfix-2.1.x`.

closed 2024-03-28 by thisiskeithb

thisiskeithb, 2024-04-27:
Patch for `2.1.x` submitted:
- https://github.com/MarlinFirmware/Marlin/pull/27021