MarlinFirmware/Marlin #26861: [BUG] (bug summary)s-curve deceleration Lost step

zhuxiufei, 2024-03-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I tested the S-curve. In the deceleration stage, the last deceleration step could not reach the set end speed because of the accuracy of the floating point number conversion. (The cumulative value of deceleration time cannot reach the total deceleration time, but the number of deceleration pulses is reached)

### Version of Marlin Firmware

marlin 2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-03-11:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

Also, please attach your configs.

ellensp, 2024-03-11:
And fill out "Steps to Reproduce" in detail.

zhuxiufei, 2024-03-11:
[run_logs.zip](https://github.com/MarlinFirmware/Marlin/files/14556452/run_logs.zip)
The S-curve code was stripped, the test could run in linux or arm, and the deceleration was actually detected complete, but did not reach the final_rate problem

zhuxiufei, 2024-03-11:
If I increase STEPPER_TIMER_RATE, things will improve.  I suspect that calc_timer_interval will have a bias in the calculation and that this bias will accumulate

InsanityAutomation, 2024-04-07:
Can you see if the change in #26881 makes any impact to this, or if we are chasing a separate issue?

closed 2024-08-29 by github-actions[bot]