MarlinFirmware/Marlin #26862: [BUG] BOTH and EITHER definitions missing from `macros.h`

xaviervilla, 2024-03-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Using VSCode with Auto Build Marlin, all builds I attempt (including default files/configs) fail due to configuration.h files containing binary operators that are not defined. It appears EITHER and BOTH used to exist in the macros.h file but they are no longer there.

### Expected behavior

I expected autobuild to succeed. 

### Actual behavior

``` 
> platformio run --silent -e melzi_optiboot_optimized
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:39:0,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:284:11: error: missing binary operator before token "(" 
 #if EITHER(PARKING_EXTRUDER, MAGNETIC_PARKING_EXTRUDER)
           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:987:11: error: missing binary operator before token "(" 
 #if EITHER(MORGAN_SCARA, MP_SCARA)
           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:1788:11: error: missing binary operator before token "("
 #if EITHER(MIN_SOFTWARE_ENDSTOPS, MAX_SOFTWARE_ENDSTOPS)
           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:1990:11: error: missing binary operator before token "("
 #if EITHER(AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_BILINEAR)
           ^
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:39:0,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:3036:11: error: missing binary operator before token "("
 #if EITHER(ANYCUBIC_LCD_I3MEGA, ANYCUBIC_LCD_CHIRON)
           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration.h:3324:11: error: missing binary operator before token "("
 #if EITHER(RGB_LED, RGBW_LED)
           ^
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:56:0,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:305:11: error: missing binary operator before token "("
   #if BOTH(ADAPTIVE_FAN_SLOWING, PIDTEMP)
           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1089:11: error: missing binary operator before token "("
 #if EITHER(INPUT_SHAPING_X, INPUT_SHAPING_Y)
           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1325:11: error: missing binary operator before token "("
 #if EITHER(DIGIPOT_MCP4018, DIGIPOT_MCP4451)
           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1382:28: error: missing binary operator before token "("
 #if HAS_BED_PROBE && EITHER(HAS_MARLINUI_MENU, HAS_TFT_LVGL_UI)
                            ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1437:11: error: missing binary operator before token "("
 #if EITHER(HAS_DISPLAY, DWIN_LCD_PROUI)
           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:1501:26: error: missing binary operator before token "("
 #if HAS_DISPLAY && EITHER(SDSUPPORT, SET_PROGRESS_MANUALLY)
                          ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2149:11: error: missing binary operator before token "("
 #if EITHER(AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_UBL)
           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2184:11: error: missing binary operator before token "("
 #if EITHER(MESH_BED_LEVELING, AUTO_BED_LEVELING_UBL)
           ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2192:9: error: missing binary operator before token "("
 #if BOTH(AUTO_BED_LEVELING_UBL, EEPROM_SETTINGS)
         ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2306:11: error: missing binary operator before token "("
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:2394:9: error: missing binary operator before token "("
 #if BOTH(SDSUPPORT, DIRECT_STEPPING)
         ^
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../../Configuration_adv.h:3336:11: error: missing binary operator before token "("
 #if EITHER(SPINDLE_FEATURE, LASER_FEATURE)
           ^
Error: Failed to parse Marlin features. See previous error messages.
========================================================================= [FAILED] Took 2.05 seconds =========================================================================

Environment               Status    Duration
------------------------  --------  ------------
melzi_optiboot_optimized  FAILED    00:00:02.048
=================================================================== 1 failed, 0 succeeded in 00:00:02.048 =================================================================== 
```

### Steps to Reproduce

Pull latest Marlin-bugfix-2.1.x branch, open the folder in VScode, click autobuild marlin. 
FIrst noticed on 9a1c993719b30dabfb2ab62101ba249dffa459c7 but I see that it still happens going back one month.

### Version of Marlin Firmware

2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/14554621/Marlin.zip)

ellensp, 2024-03-11:
They are not missing they where deliberately removed

Use correct Configs for your code version.

closed 2024-03-11 by ellensp

ellensp, 2024-03-11:
Configs say 2.1.2.2, but that version of code has those macros 
BOTH https://github.com/MarlinFirmware/Marlin/blob/2.1.2.2/Marlin/src/core/macros.h#L279 
EITHER https://github.com/MarlinFirmware/Marlin/blob/2.1.2.2/Marlin/src/core/macros.h#L280

ellensp, 2024-03-11:
Removed here https://github.com/MarlinFirmware/Marlin/pull/25908/files