MarlinFirmware/Marlin #26863: Integrating New Custom G-Codes (M9910-M9919) into Marlin 2.1: Undefined Reference Error

Elias-Ta, 2024-03-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm encountering an issue while attempting to add new custom G-codes to Marlin 2.1.* firmware. Specifically, I'm aiming to introduce G-codes M9910 to M9919 into the firmware. While my implementation method has worked in Marlin 2.0., I'm facing difficulties integrating it into Marlin 2.1..

Here's the approach I've taken:

1. Configuration_adv.h:

``` 
#define New_Custom_Gcodes
```

2. gcode.cpp:

``` 
// Inside GcodeSuite::process_parsed_command()
switch (parser.command_letter) {
    ....
    case 'M':
        switch (parser.codenum) {
            // Existing cases...
            #if ENABLED(New_Custom_Gcodes)
                case 9910:
                case 9911:
                case 9912:
                case 9913:
                case 9914:
                case 9915:
                case 9916:
                case 9917:
                case 9918:
                case 9919:
                    M9910_M9919(); // Define/execute G-code macro
                    break;
            #endif
            // Existing cases...
        }
    // Rest of the function...
}
```

3. gcode.h:

``` 
class GcodeSuite {
public:
    // Existing codes...
    #if ENABLED(New_Custom_Gcodes)
        static void M9910_M9919();
    #endif
};
```

4. I've created a new file named M9910_M9919.cpp under Marlin\src\gcode\control, M9910_M9919.cpp:
```
#include "../../inc/MarlinConfig.h"

#if ENABLED(New_Custom_Gcodes)

#include "../gcode.h"
#include "../queue.h"
#include "../parser.h"

void GcodeSuite::M9910_M9919() {
    // Execute Custom G-codes
    switch (parser.codenum - 9910) {
        case 0:
            // Do something
            break;
        case 1:
            // Do something
            break;
        // Cases for M9911, M9912, ..., M9919
        default:
            break;
    }
}

#endif

```

Despite these steps, I'm encountering the following error during the linking stage:

```
c:/users/****/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/STM32H723ZE_btt/src/src/gcode/gcode.cpp.o: in function `GcodeSuite::process_parsed_command(bool)':
C:\Users\***\***\***\***\Marlin-bugfix-2.1.x/Marlin\src\gcode/gcode.cpp:1123: undefined reference to `GcodeSuite::M9910_M9919()'
collect2.exe: error: ld returned 1 exit status

```

I understand that This error suggests that the linker cannot find a reference to the M9910_M9919 function, indicating a problem with its definition or how it's being referenced, however I couldn't find what is the mistake in this case.

I'd greatly appreciate any insights or suggestions on resolving this issue. If there's anything I've overlooked or if there are specific changes in Marlin 2.1.* that may affect how custom G-codes are integrated, please let me know. Thank you in advance for your assistance!

### Version of Marlin Firmware

bugfix 2.1.*

### Electronics

BOARD BTT OCTOPUS PRO V1.1 STM32H723ZE*

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-03-11:
your probably being caught by source filtering
features.ini tells platformio what  code to compile and what to ignore

to confirm simply build and log the build, see if your M9910_M9919.cpp is listed 

I suspect it will not, you need to add a line like

`New_Custom_Gcodes   =  build_src_filter=+<src/gcode/control/M9910_M9919.cpp>`

I do not know if this can be mixed case..  we stick to uppercase defines...

Elias-Ta, 2024-03-11:
@ellensp thank you for your help,  it's working and compiling now

ellensp, 2024-03-11:
Are you aware we already have GCODE_MACROS? 

/**
 * G-code Macros
 *
 * Add G-codes M810-M819 to define and run G-code macros.
 * Macros are not saved to EEPROM.
 */
//#define GCODE_MACROS

Elias-Ta, 2024-03-11:
yeah I'm aware about that, however I think those macros are defined using Gcodes (M810-M819) during run time, what I am trying to do is use some functions in marlin to control an 8 axis machine (part of it is a 3d printer), and therfore I need some predefined codes that are just a combination of other already-exsisted gcodes, just to make things easier to fix and change later on marlin side rather than the other host software side