MarlinFirmware/Marlin #26865: [BUG] Z-increases twice when homing after Marlin boot

Sophist-UK, 2024-03-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

On first home after boot, head does the following:

1. Move upwards
2. Home X & Y
3. Move upwards
4. Home Z

In `Configuration.h` the line `//#define Z_CLEARANCE_FOR_HOMING  4` is still commented out, so I assume that the movement for steps 1 and 3 above is this default of 4mm.

The issue is that it is doing this twice - and that this uses up more than the amount of Z-space I have allowed for between software Z-max (where the head might end up before power off) and the actual Z-max when the head hits the top gantry.

I understand why it does step 1. above, as it needs to ensure that the head doesn't scratch the (possibly uneven) bed and / or avoid clamps. But I do not understand why it does it a second time.

### Bug Timeline

No idea.

### Expected behavior

When it has no idea of the head position i.e. after boot, I expect it to do this z-hop only once on a Home.

### Actual behavior

Head does a double z-hop.

### Steps to Reproduce

1. Turn on printer.
2. Issue G28.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Dagoma DE200

### Electronics

MKS Base 1.5 AVR

### LCD/Controller

REPRAPDISCOUNT_FULL_GRAPHICS

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Temp.zip](https://github.com/MarlinFirmware/Marlin/files/14564942/Temp.zip)

Sophist-UK, 2024-03-11:
`Z_CLEARANCE_BETWEEN_PROBES` is 5 so it is probably using that.

thisiskeithb, 2024-03-11:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as [Pronterface](https://github.com/kliment/Printrun/releases/latest), [Repetier Host](https://www.repetier.com/download-now/), or [OctoPrint](https://octoprint.org/download/).
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Sophist-UK, 2024-03-11:
@thisiskeithb I will do this tomorrow (assuming I have the time) as it is late here now.

I have read the `G28.cpp` and `motion.cpp` code and cannot immediately see any bug in this. The first z-hop is applied at https://github.com/MarlinFirmware/Marlin/blob/9a1c993719b30dabfb2ab62101ba249dffa459c7/Marlin/src/gcode/calibrate/G28.cpp#L430, then X and Y are homed, then `home_z_safely` is called at https://github.com/MarlinFirmware/Marlin/blob/9a1c993719b30dabfb2ab62101ba249dffa459c7/Marlin/src/gcode/calibrate/G28.cpp#L510 which then moves to the SafeXY and eventually calls `homeaxis(Z_AXIS)` at https://github.com/MarlinFirmware/Marlin/blob/9a1c993719b30dabfb2ab62101ba249dffa459c7/Marlin/src/gcode/calibrate/G28.cpp#L137, and `homeaxis` is defined at https://github.com/MarlinFirmware/Marlin/blob/9a1c993719b30dabfb2ab62101ba249dffa459c7/Marlin/src/module/motion.cpp#L2094.

The comments in `motion.cpp` do talk about a z-hop, but neither `SENSORLESS_BACKOFF_MM` or `XY_COUNTERPART_BACKOFF_MM` is defined so we shouldn't do any Z-axis backoffs here and then we should do a home move at https://github.com/MarlinFirmware/Marlin/blob/9a1c993719b30dabfb2ab62101ba249dffa459c7/Marlin/src/module/motion.cpp#L2173. 

But let's try the debug version tomorrow and see what output we get.

Sophist-UK, 2024-03-12:
Well, I had set the following:
```
#define Z_CLEARANCE_FOR_HOMING       5 // Minimum for stock probe is 5mm (to allow for bed skew cf. gantry / Y guide rods
#define Z_CLEARANCE_DEPLOY_PROBE   10 // (mm) Z Clearance for Deploy/Stow
```
And I expect it was this mismatch that was causing an extra 5mm z-hop. I will test this fix with my next impending build.

But here is the octoprint serial log just in case:
[serial.log](https://github.com/MarlinFirmware/Marlin/files/14578993/serial.log)

Sophist-UK, 2024-03-15:
Ok - I made both of the above values the same, and the 2nd z-hop is less but still exists.

Hopefully the serial.log I previously loaded will help diagnose the problem, but if not I can rerun it with the current settings.

Just so that we are clear about why this is an issue (for a bed-slinger - slightly different issues apply if it is e.g. a COREXY):

1. You need to set both these values so that if the head is resting on the bed when you boot the firmware, you increase the Z-height sufficient so as to avoid any clamps, bed slopes etc. and not to scrape the bed.
2. If the head is at Z-MAX_POS when you boot the firmware, the total of the two z-hops may exceed the Z_CLEARANCE_FOR_HOMING clearance that the firmware comments require you to leave and the head may grind.

Sophist-UK, 2024-03-19:
One more thing I noticed (but cannot track down in the code) is that I have `HOMING_BUMP_MM` set to `{ 3, 3, 1 }` so the z-bump should be 1mm, but it looks to me like it is doing a 5mm hop before doing a second homing z-probe.

closed 2024-07-02 by github-actions[bot]