MarlinFirmware/Marlin #26866: [BUG] M73 no percentage progress when printing via OctoPrint

lethuer, 2024-03-12:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Hi ! I'm using latest Marlin 2.1.2.1 together with latest TFT firmware (tft_fw_1.27.x_patch_11) from Digant on my Artillery Genius (not Pro).
https://www.thingiverse.com/thing:4294049#Marlin-Firmware

My problem is that the percentage progress bar is only working if I'm printing via usb stick. But my normal procedure is using octoprint, then the bar and value always showing 0%. If I stop a print or a print is done then 100% is displayed.

I've checked the firmware and "SET_PROGRESS_MANUALLY" and "M73_REPORT" are defined, "SDSUPPORT" is not.
Also I've checked that M73 plugin is installed to octoprint:
https://plugins.octoprint.org/plugins/m73progress/

Can anybody help ? I'm searching since days...

[image]
[image]
[image]
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/14565756/Config.zip)

### Bug Timeline

2.1.2.1

### Expected behavior

Percentage value should be displayed

### Actual behavior

No percentage value / bar

### Version of Marlin Firmware

2.1.2.1

### Printer model

Artillery Genius (not Pro)

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-03-12:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

InsanityAutomation, 2024-03-13:
Can confirm that old gen MKS TFT if very basic in what it supports. IIRC it only reports progress on what it runs locally.

lethuer, 2024-03-15:
> Can confirm that old gen MKS TFT if very basic in what it supports. IIRC it only reports progress on what it runs locally.

What do you mean by "locally" ?

If I use terminal from octoprint I can observe this:
```
Send: M118 P0 A1 action:print_start
Recv: //action:print_start
Recv: ok P15 B3
[...]
Send: M73 P25 R43
Recv: echo: M73 Progress: 25%; Time left: 43m;
Recv: ok P15 B3
```

So my interpretation of this is that the firmware reacts properly because there is a "Recv" message ?
In this case the tft software wouldn't handle this correct ?
If I wouldn't define "SET_PROGRESS_MANUALLY" would then Marlin again calculate the progress and send this to the tft ?

InsanityAutomation, 2024-03-15:
That shows Marlin is behaving properly. Its a limitation on the TFT side not displaying it.

closed 2024-03-15 by thisiskeithb