MarlinFirmware/Marlin #26868: Homing Sequence with extra axis

Elias-Ta, 2024-03-14:
### Describe the feature you want

Hello, I'm grappling with a complex setup involving a 3-axis 3D printer with an additional 5-axis component. I'm experimenting with post-processing tasks such as part removal and other functions using these extra axes. My question is, is there a method to synchronize the homing process for these two additional axes? I'm seeking a solution akin to QUICK_HOME for the X and Y axes.

Previously, I attempted to integrate new functions into G28.cpp, motion.cpp and motion.h, following a similar approach to QUICK_HOME. However, it appears that Marlin controls the extra axes differently, resulting in unsuccessful outcomes.

DerAndere1, 2024-04-22:
I have this Feature (QUICK_HOME_ALL_NON_Z_AXES) in my fork. I will prepare a pull request this week so you can see what changes are required.

Elias-Ta, 2024-04-23:
@DerAndere1 Thank you for your answer, I'm looking forward to your pull request.
BTW, your fork has great features, and it seems great for my next prototype.

DerAndere1, 2024-04-23:
@Elias-Ta : Please let us know if the changes proposed in PR #27005 work for you

closed 2024-08-27 by thisiskeithb

thisiskeithb, 2024-08-27:
Closing because:

> This PR is more a show case of what changes are necessary. I do not encourage to merge this, because there is hardly a single homing sequence that works for every multi-axis machine. For such complex machines, a custom-tailored safe homing sequence should be implemented by the machine builder.

> I still dont recommend to merge this for the reasons explained in the description of this PR.

_Originally posted by @DerAndere1 in https://github.com/MarlinFirmware/Marlin/issues/27005#issuecomment-2313563411_