MarlinFirmware/Marlin #26870: [BUG] 2.1.2.2 stuck on boot with SKR 1.4 (LPC1768)

BaBBa71, 2024-03-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have the following startup issue with the latest release 2.1.2.2, both with the "Current Release" version and also when trying with the "Marlin 2.1.2.2 with bug fixes" version. I have rewritten all the parameters that I currently use in version 2.1.2.1 in the configuration files of the aforementioned release. However, after compiling without any errors and loading the file onto the SD card, it remains stuck at the Marlin logo during boot with the release, but does not proceed with the loading.

I conducted further testing by using the new files in the old release 2.1.2.1, modifying only the two values of #define CONFIGURATION_H_VERSION respectively in Configuration.h and Configuration_adv.h with the previous version. I compile without errors, and the system proceeds smoothly with the boot. I believe there might be some bug in the current release/configuration parameter when using the SKR 1.4 (non-turbo) and compiling for the LPC1768 microcontroller.

### Expected behavior

After a compilation without any warnings, errors, or indications for the LPC1768 chip of the BIGTREETECH SKR V1.4, a complete system boot occurs.

### Actual behavior

Stuck on Marlin boot logo.

### Steps to Reproduce

Use configuration files with custom boot logo on a SKR 1.4

### Version of Marlin Firmware

2.1.2.2

### Printer model

Custom Cartesian 

### Electronics

BIGTREETECH SKR V1.4 with TMC2130 on XY and 2209 on ZE

### LCD/Controller

FYSETC_MINI_12864_2_1

### Other add-ons

BL-Touch

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configurations_file.zip](https://github.com/MarlinFirmware/Marlin/files/14606177/configurations_file.zip)
[custom_logo.zip](https://github.com/MarlinFirmware/Marlin/files/14606247/custom_logo.zip)

thisiskeithb, 2024-03-14:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/26811

thisiskeithb, 2024-03-14:
This issue is already fixed in [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip).

closed 2024-03-14 by thisiskeithb

BaBBa71, 2024-03-14:
@thisiskeithb Thank you for the fast feedback. I performed the test with https://github.com/MarlinFirmware/Marlin/archive/2.1.x.zip and not with the nightly build https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip, which I also tried to compile. 
However, with the same 2.1.2.2 files, it doesn't work, as the configuration files are likely different since they refer to version 2.1.3. I'll wait for a stable release or fixed 2.1.2.2 release. Thanks again.

thisiskeithb, 2024-03-14:
> However, with the same 2.1.2.2 files, it doesn't work, as the configuration files are likely different since they refer to version 2.1.3

Yep. You'll need to port your settings over. I recommend sticking with `2.1.2.1` or moving to `bugfx-2.1.x` since there are several bugs in `2.1.2.2`.