MarlinFirmware/Marlin #26872: Z Probe Offset Issue with Bltouch

ve3sjk, 2024-03-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I just upgraded my machine to the latest code and did the standard Z probe offset test with G28, M211 S0, then using the reprap discount full graphics controller steppind down the z axis to a piece of paper.

I have the z probe offset wizard disabled in the firmware.

That test gave me a z probe offset of -0.70 where the paper was snug on a clean nozzle

When i add the value and print the first layer is in the air despite the paper being very snug in the test.

I got the printer working by using my old offset that was -1.62 mm

I cannot figure out why proper z probe offset test is that much off from what i get on the display when i test it with paper so it was telling it was -0.70 mm when it was in fact -1.60 mm. This test previously always resulted in somewhere near z-1.60 with the bugfix branch i had in the printer for years.

[Marlin Config.zip](https://github.com/MarlinFirmware/Marlin/files/14609820/Marlin.Config.zip)

### Bug Timeline

unknown

### Expected behavior

z probe offset test should result in a reading of -1.62 which is the actual true offset

### Actual behavior

z probe offset test reads -0.70 mm

### Steps to Reproduce

Perform Z probe offset test

### Version of Marlin Firmware

Marlin 2.1.2.2

### Printer model

Anet A8 with E3DV6 extruder

### Electronics

MKS Base V1.2

### LCD/Controller

Reprap discount full graphics controller

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-03-15:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

rondlh, 2024-03-15:
> That test gave me a z probe offset of -0.70 where the paper was snug on a clean nozzle

Did you have the bed and nozzle at print temperature?

ve3sjk, 2024-03-15:
It turns out it was operator error, the ticket can be closed.

I did not ensure that the z offset was zero before i did the procedure so when I homed the axis it was including the default that i had setup in the firmware, i think i am going to default the offset in my files and calculate on first run.

closed 2024-03-15 by thisiskeithb