MarlinFirmware/Marlin #26877: [BUG] Sensorless Homing not functioning CoreXY

theomajigga, 2024-03-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

CoreXY Hypercube style printer FLYF407ZG latest marlin. Dual Z screws. Pins updated for UART and Z2 axis on E0. Wires ran from endstops on board to pin on tmc2209. 

Currently have no belts on  it to see if there was binding, still does the same thing.

Have attempted every value for threshold that I can think of, changed bump off, etc. 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/14621475/Configuration.zip)

### Expected behavior

Homing sequence to finish properly. 

### Actual behavior

Bounces off of axis slightly and returns to original spot. 

### Steps to Reproduce

Write firmware to card, and run homing sequence. 

### Version of Marlin Firmware

Marlin 2.1.2.2

### Printer model

Hypercube Evolution

### Electronics

FLYF407ZG

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-03-16:
You really should not use Sensorless for any axis using a lead screw, especially Z. 
Sensorless is not at all accurate or repeatable. on Z this will result in damaging your bed.

You need to tune Sensorless to your hardware, not only STALL_SENSITIVITY but also the homing feed rate.

So called Sensorless is a misnomer, in reality it is using the stepper drivers stall detection feature. As such you need the axis to be moving fast enough and to stop abruptly enough to trigger the stepper driver stall detection.

Finding the correct setting for your machine takes time and experimentation.
It also also much harder to tune on a core xy, since the two stepper motors are joined via the belt. 

Also test bugfix2.1.x   Many things are broken on 2.1.2.2

vfbank, 2024-03-16:
Sensorless homing does not work properly with the leadscrew Z method. The TMC2209 operates by detecting bump stall (impact), so it does not operate properly on leadscrews with couplers connected.

theomajigga, 2024-03-16:
Adding it to Z was a last ditch effort to adding it to any axis with a TMC on it. 

Will do bugfix M122 does deliver output.

Wil test bugfix.

PKav89, 2024-05-17:
Hi!
Same issue.
Sensorless homing on latest bugfix works on cartesian machines, but not on CoreXY.
2.1.2.2 works well on CoreXY.
Have you found solution?

closed 2024-08-30 by github-actions[bot]