MarlinFirmware/Marlin #26878: [BUG] Dual X Carriage Unable to Park

nichols89ben, 2024-03-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Having issues with DUAL_X_CARRIAGE and being unable to auto park or park X2 at all. The firmware settings look correct based on what I read in the documentation. See attached video links for easy clarification of the issue. But when homing, te X2 head moves as if it will park but only moves to the left and not back to the endstop. Also selecting certain autopark/full control modes causes it to crash into the X1 Carriage. The X2 carriage is responsive in the UI and will move fine manually, but homing and parking is the issue. I attached videos, configs and pins.h. These are also what the endstop pins show when in debugging. Any help is appreciated. 
Pin		
PG6	T0 Xstop	- X_MIN:0 when hit X_MIN:1 when removed
PG9	Ystop	- Y_MIN:0 when hit Y_MIN:1 when removed
PG10	T1 X2Stop - Z_MIN:0 when hit Z_MIN:1 when removed 

Configs:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/14622350/config.zip)

Videos:
https://drive.google.com/file/d/11NCRGhEjbzqHYfuVxzfL01xzpnt_FIoE/view?usp=sharing
https://drive.google.com/file/d/1q-FJaFxrjhutsMmYiXYODnDMw1kP9xHs/view?usp=sharing

### Expected behavior

X2 carriage to park when homed

### Version of Marlin Firmware

bugfix-2.1.x 2024-02-02

### Printer model

SV04

### Electronics

BTT Octopus

### LCD/Controller

BTT

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-03-16:
PG10 is also Z Z_DIAG_PIN,  did you remove DIAG_JUMPERS for Z axis?

ellensp, 2024-03-16:
"PG10 T1 X2Stop - Z_MIN:0 when hit Z_MIN:1 when removed"

huh?

you have set X2_DIAG_PIN PG10
but then you set   #define Z_STOP_PIN                  X2_DIAG_PIN  // Z-STOP

so z is using PG10 not X2

X2 is using  E0_DIAG_PIN

closed 2024-03-16 by ellensp

ellensp, 2024-03-16:
In Configuration_advanced you have 
    #define X2_STOP_PIN X2_DIAG_PIN    //#BEN_ENDSTOP // X2 endstop pin override
But this line is not used when you have DUAL_X_CARRIAGE

With  DUAL_X_CARRIAGE 
T0 homes to X-MIN  ie X_DIAG_PIN (PG6)
T1 homes to X-MAX ie E0_DIAG_PIN (PG12)

most of the changes you made to Marlin/src/pins/stm32f4/pins_BTT_OCTOPUS_V1_common.h are not needed
undo all  but these
```
#define X2_STEP_PIN                         PE2   // MOTOR 6
#define X2_DIR_PIN                          PE3
#define X2_ENABLE_PIN                       PD4
#ifndef X2_CS_PIN

  #define X2_CS_PIN                         PE1
#endif

  #define X2_SERIAL_TX_PIN                  PE1
  #define X2_SERIAL_RX_PIN      X2_SERIAL_TX_PIN
```

ellensp, 2024-03-16:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

nichols89ben, 2024-03-16:
> ### Configuration_advanced.h:
> 
> In Configuration_advanced.h you have ` #define X2_STOP_PIN X2_DIAG_PIN //#BEN_ENDSTOP // X2 endstop pin override` But this block of the config file is not used when you have DUAL_X_CARRIAGE, so all these setting are ignored.
> 
> ```
> #if HAS_X2_STEPPER && DISABLED(DUAL_X_CARRIAGE)
>   //#define INVERT_X2_VS_X_DIR        // X2 direction signal is the opposite of X
>   #define X_DUAL_ENDSTOPS           //#BEN_ENDSTOP // X2 has its own endstop
>   #if ENABLED(X_DUAL_ENDSTOPS)
>     //#define X2_STOP_PIN X_MAX_PIN   //#BEN_ENDSTOP // X2 endstop pin override
>     #define X2_STOP_PIN X2_DIAG_PIN    //#BEN_ENDSTOP // X2 endstop pin override
>     #define X2_ENDSTOP_ADJUSTMENT  0  // X2 offset relative to X endstop
>   #endif
> #endif
> ```
> 
> HAS_X2_STEPPER is for when you have two X stepper drivers on the on X axis, like dual Z, but for X. These are not the setting you want for DUAL_X_CARRIAGE
> # NOTE:
> 
> With DUAL_X_CARRIAGE T0 homes to X-MIN ie X_DIAG_PIN (PG6) T1 homes to X-MAX ie E0_DIAG_PIN (PG12)
> ### pins_BTT_OCTOPUS_V1_common:
> 
> Most of the changes you made to Marlin/src/pins/stm32f4/pins_BTT_OCTOPUS_V1_common.h are not needed undo all but these
> 
> ```
> #define X2_STEP_PIN                         PE2   // MOTOR 6
> #define X2_DIR_PIN                          PE3
> #define X2_ENABLE_PIN                       PD4
> #ifndef X2_CS_PIN
> 
>   #define X2_CS_PIN                         PE1
> #endif
> 
>   #define X2_SERIAL_TX_PIN                  PE1
>   #define X2_SERIAL_RX_PIN      X2_SERIAL_TX_PIN
> ```
> 
> Even these are not technically needed as marlin will automatically use E2 as X2 and will warn you with #warning "Note: Auto-assigned X2 STEP/DIR/ENABLE_PINs to unused En_STEP/DIR/ENABLE_PINs. (Define NO_AUTO_ASSIGN_WARNING to suppress this warning.)"

Hi, thank you for the detailed reply. I reset the pins back to default as you suggested and re flashed, however the behavior is still the same as depicted prior. I made sure to reset to defaults with M502 and M500. 

I reached out to the discord threads several times and never got a response, ive been at it for weeks and the bug report was my last resort. Any other suggestions is appreciated

nichols89ben, 2024-03-16:
> ### Configuration_advanced.h:
>
> In Configuration_advanced.h you have ` #define X2_STOP_PIN X2_DIAG_PIN //#BEN_ENDSTOP // X2 endstop pin override` But this block of the config file is not used when [...]

Also, I know you mentioned Marlin will automatically use E2 as X2, I dont know if that applies to the endstop also. The Stepper is in the E2 defined slot, but In the pins file E2 endstop is defined as PG14 and a runout sensor. I have the X2 endstop plugged into #define Z_DIAG_PIN  PG10  // Z-STOP. The pin debug also reports  Z_MIN pin

[image]

ellensp, 2024-03-16:
stepper driver pins have little to do with endstops. endstops pin don't change.

I repeat:

With DUAL_X_CARRIAGE
Tool 0 homes to X-MIN ie X_DIAG_PIN (PG6)
Tool 1 homes to X-MAX ie E0_DIAG_PIN (PG12)

nichols89ben, 2024-03-16:
> stepper driver pins have little to do with endstops. endstops pin don't change.
> 
> I repeat:
> 
> With DUAL_X_CARRIAGE Tool 0 homes to X-MIN ie X_DIAG_PIN (PG6) Tool 1 homes to X-MAX ie E0_DIAG_PIN (PG12)

Thanks again, that did fix the behavior. I didnt try that earlier because the pins.h specifically refer to those four as filament run out sensors.  I previously had a run out sensor in that pin but removed it and shifted everything down one. Will marlin pick up the new run out sensor locations or does that need to be specified?

[image]