MarlinFirmware/Marlin #26885: [BUG] Firmware Builder: Tramming Wizard > Front-Right > Measure and Back-Right > Measure don't respond

objecttothis, 2024-03-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm trying to complete the tramming Wizard.  I am able to successfully take measurements in Back-Left and Front-Left, but Back-Right and Front-Right do not work.

### Bug Timeline

New

### Expected behavior

I expect clicking Measure to take measurement like it does for Back-Left and Front-Left.

### Actual behavior

Clicking Tramming Wizard > Front-Right > Measure after taking initial measurement from Front-Left does not respond.  If I start with Origin of Front-Right the printer centers the nozzle and raises the Z-axis but then does nothing further.  Same for Back-Right.

### Steps to Reproduce

1. Firmware Builder (ID 92902ff4)
2. Flash Firmware
3. Configure Probe offsets (X -77, Y -20, Z -0.11)
4. Motion > Tramming Wizard > Front-Left > Measure
5. Motion > Tramming Wizard > Front-Right > Measure (or Back-Right > Measure)

### Version of Marlin Firmware

2.1.x-bugfix

### Printer model

Ender-3

### Electronics

4.2.7, CR-touch, Micro-swiss DD w/all metal hotend

### LCD/Controller

Stock

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I don't know how to extract the Configuration.h and Configuration_adv.h, from the Firmware builder .bin file.  If someone posts a link to instructions I will upload this. Build ID: 92902ff4

ellensp, 2024-03-18:
Please attach your configuration files.

objecttothis, 2024-03-18:
> Please attach your configuration files.

As I stated in the bug report, I downloaded the firmware from the Firmware Builder and I don't know how to extract the Configuration.h and Configuration_adv.h files from the bin provided.  If there is a way for me to get the config files the firmware builder used, then let me know any I will provide them.

objecttothis, 2024-03-18:
https://marlin.crc.id.au/builder/

For some reason I thought this a Marlin project. I'll reach out to them to figure out how to get the config files.

closed 2024-03-18 by thisiskeithb