MarlinFirmware/Marlin #26887: [FR] Z Self Calibration

nickweedon, 2024-03-18:
### Is your feature request related to a problem? Please describe.

The problem that this PR would address is that every time a nozzle is changed or a Z homing probe is innacurate then the nozzle will not be at the correct Z position and may even scrape against the bed or cause the extruded to click or possibly even jam (i.e. if using TPU filament).

### Are you looking for hardware support?

Ideally something like the Voron Sex Bolt (yes that is what this thing is called sadly) or an equivalent knockoff/clone.

### Describe the feature you want

The ability to measure the Z offset automatically using a Z end stop switch attached to the bed.
This idea is best described by this already existing implementation for Klipper here: https://github.com/protoloft/klipper_z_calibration/wiki/What-It-Does

### Additional context

I saw already that something similar to this was already proposed here https://github.com/MarlinFirmware/Marlin/issues/25657 but I would like to call out that support for multiple tool offsets does not address the problem described here. Moreover, if it did then it would be unlikely that both hardware and software solutions would have been produced already by Voron and Klipper.

I may actually be interested in working on this PR myself if there is no push back here (20+ year dev with experience in embedded systems including autonomous vehicles)

studiodyne, 2024-04-06:
It exists , i use it, but you need to be electronician or very geek.
The luzlbolt machine with two switching extruders , calibrate the machine with the marlin embeded calibration 
I have made my own switching machine and i use it
The difficulty is to ensure electric conduction of the nozzle if you use a metal part
But you can a switch if it is just for z. But take care if nozzle is cold or dirty , the z can be dramatically wrong

Add a z switch somewhere , and use calibration when changing nozzle 
I have it , it works fine
I have tweaked it to avoir tool swapping , because calibration swap 4 times between extruders

DerAndere1, 2024-04-08:
To elaborate on the comment by studiodyne: The implementation for RRF (https://github.com/pRINTERnOODLE/Auto-Z-calibration-for-RRF-3.3-or-later-and-Klicky-Probe) seemingly just auto-adjusts the tool-length offset (hotend offset). To achieve this with Marlin, I suggest the following (untested) procedure: configure your actual extruder as Extruder 1. Tool 0 (E0) should be a dummy, not used for printing. It only serves as a reference to have a constant nozzle-to-probe offset. Enable `HOTEND_OFFSETS`, `CALIBRATION_GCODE` and `CNC_COORDINATE_SYSTEMS`. HOTEND_OFFSET for tool 0 must be 0mm. Configure the tool-setter as the calibration object. If the tool-setter can only measure the tool-length, disable all of CALIBRATION_MEASURE_RIGHT, CALIBRATION_MEASURE_LEFT, CALIBRATION_MEASURE_FRONT, CALIBRATION_MEASURE_BACK. Connect and configure your z-probe. Connect the tool-setter. The pin that is connected to the tool-setter's signal needs to be defined as `CALIBRATION_PIN`. Set your nozzle-to-probe offset (it should be constant for a given probe). Then, run this is this G-code:
```
G28;  Home all axes.
G55;  Select workspace coordinate system 2
G1 X... Y... ;  move above the fixture / print-bed
T0; switch to reference tool
G38.2;  probe towards the fixture/ print-bed, stop when probe triggers 
G92 Z<nozzle-to-probe-offset>;  G10 is not supported. use G92 to set the Z-position in the current workspace to the nozzle-to-probe offset. nozzle-to-probe-offset is the offset between probe and reference tool 0!!!!
T1;  Switch to Extruder 1.
G425 T1; Calibrate tool 1. This adjusts the hotend offset (tool length offset) for hotend 1
```

DerAndere1, 2024-04-09:
In addition to the above, if you want to use the tool-setter also to measure the nozzle-to-probe Z offset (not recommended because it disregards the trigger-point of the z-probe), configure the probe as an additional dummy tool (extruder 3). Measure and set the nozzle-to-probe offset as good as possible using an alternative measurement method. Set the hotend offset for tool 3 to the same value as the nozzle-to-probe offset. Then run the following G-code: 

```
T0; switch to reference tool
G28;  Home all axes.
M218 T1 Z<tool_length>;  G10 is not supported. Use M218 to roughly set the hotend offset. It must be within 5 mm of the true nozzle length. this is a limitation of the current implementation of G425.
G425;  Calibrate all tools. This adjusts backlash compensation and the hotend offset (tool length offset) for all tools
M218;  Report hotend offsets. Note down the hotend Z offset for tool 3
M851 Z<hotend_offset_T3>;  Set the nozzle-to-probe Z offset equal to the new hotend Z offset for tool 3
G55;  Select workspace coordinate system 2
G1 X... Y... Z...;  move above the fixture / print-bed
G38.2;  probe towards the fixture / print-bed, stop when probe triggers 
G92 Z<-(nozzle-to-probe-offset)>;  G10 is not supported. use G92 to set the Z-position in the current workspace to the negative of the nozzle-to-probe offset. nozzle-to-probe-offset is the offset between probe and reference tool 0!!!!
T1;  Switch to Extruder 1.
M500;  Save settings to EEPROM
```