MarlinFirmware/Marlin #26889: [BUG] Enabling I_AXIS (as alternative to E_AXIS on MiniRambo board) fails compiling caused by missing I_MS3_PIN.

hm2dev, 2024-03-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When using Marlin on an MiniRambo Board, where the Extruder is used as another axis (here i-axis) by defining
#define I_STEP_PIN                          34
#define I_DIR_PIN                            43
#define I_ENABLE_PIN                     26
and
#define I_MS1_PIN                            65
#define I_MS2_PIN                            66
(basically replacing the E0 pins with I pins)
compilation fails when trying to access I_MS3 pin, which is not defined.

Changing     
https://github.com/MarlinFirmware/Marlin/blob/05896551f1fd521865e6cc056a3acc07a6bf2c0b/Marlin/src/module/stepper.cpp#L4280
to  
`#if HAS_I_MS_PINS && PIN_EXISTS(I_MS3)`
fixes this issue (at least for the i-axis).

Note, there is also a bunch of semicolons missing in the current stepper.cpp after "break" statements related to I, J, K, U, V and W axes in function Stepper::microstep_ms(...), e.g.
https://github.com/MarlinFirmware/Marlin/blob/05896551f1fd521865e6cc056a3acc07a6bf2c0b/Marlin/src/module/stepper.cpp#L4281

Adding the above pre-processor condition and fixing the semicolons makes the configuration compile and results in a working i-axis.

### Bug Timeline

Tried branch lts-2.1.1 first and then bugfix 2,1,x, Both are broken.

### Expected behavior

Should compile.

### Actual behavior

Compilation fails with undefined I_MS3_PIN.

### Steps to Reproduce

see bug details

### Version of Marlin Firmware

2.1.x

### Printer model

Custom, based on MiniRambo Board

### Electronics

MiniRambo 

### LCD/Controller

not relevant

### Other add-ons

none

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

hm2dev, 2024-03-20:
Is a Pull Request to fix this welcome, or is it to trivial for invoking this workflow?

ellensp, 2024-03-20:
PRs are welcomed and encouraged

hm2dev, 2024-03-20:
Pull requested the fix in #26891 and also added a "test case" in #26892.

thisiskeithb, 2024-03-20:
Closing since you've opened a PR. I also left a review for how configs are added/tested, so be sure to check that out.

PR #26891 (2024-03-22): Fix access to undefined _MS3 pins in stepper.cpp
Fixes #26889.
Prevent switch cases in function Stepper::microstep_ms for axis I, J, K, U, V and W from accessing x_MS3_PIN in case if x_MS3_PIN is not defined. Add missing semicolon to break statements

### Requirements

none

### Benefits

Fixes #26889.

### Configurations

none but will add a separate Pull request for features and configurations actually using and requiring this fix.

### Related Issues

Fixes #26889.