MarlinFirmware/Marlin #26890: [FR] Improved Watchdog

Sophist-UK, 2024-03-20:
### Is your feature request related to a problem? Please describe.

I created a build for AVR which did not exceed 100% SRAM (for pre-defined memory), but which didn't leave sufficient memory to run a print. The symptoms I received were a variety from the following:

1. Freezes
2. Multiple beeps
3. "SD read error" when SD card was not in use
4. Serial communications errors
5. Reboots

This was all unhelpful in the sense that there was no indication of what the problem was. For situations where the MB hung, it was unclear what was happening with hotend heating (since the display was not updating), so unclear whether we might have been having thermal runaway. Fortunately I was present when this happened and could do a manual reset to ensure that the hotend stopped heating and the fan remained on to prevent the head melting.

### Are you looking for hardware support?

No - but this issue was on an AVR.

### Describe the feature you want

1. Improved watchdog, esp. watching for Stack exhaustion.
2. Improved diagnostic information i.e. indicating the issue (and ideally indicating a solution or a URL)
3. Dying breath diagnostics (to serial and screen) + Halt rather than a reboot

InsanityAutomation, 2024-03-20:
The classic rule of thumb has always been 1k free memory for the stack, but no real way to flag that in PIO. We do have M100 memory watcher as an option, but its more a debug tool and again takes even more memory so hard to fit in when already tight and trying to watch memory...

Sophist-UK, 2024-03-20:
My current build reports `Free Memory: 1529  PlannerBufferBytes: 1600`. It prints just fine but if I go to menu Configuration when it is cooling down i.e. still printing, then it reboots for no apparent reason.

Sophist-UK, 2024-03-20:
I haven't compiled and used M100, but I suspect that it collects and analyses far more information than a watchdog would.

Ideally we would check on a regular basis (as part of the main loop perhaps) that the stack had not exhausted memory and halt the system with diagnostic info if it had.

But at a minimum IMO we should avoid rebooting and instead halt the system with whatever diagnostics we can in those circumstances where it currently reboots.

ellensp, 2024-03-20:
reboot is to ensure all heaters are off via the init routines

Sophist-UK, 2024-03-20:
Ah - well if we cannot guarantee that this can be achieved without a reboot to reset the stack, then I guess this is the only solution.

However on the previous firmware which wouldn't print, it often hung and needed a manual reset to turn off the heaters and ensure the hotend fan was on until the temp was cool enough.

InsanityAutomation, 2024-03-20:
Depends how it crashes... most platforms dont have an explicit control to say reboot or not. We call minkill to shut everything down when we can, but theres no guarantee thats occurring in the memory corruption cases above. In that case, youre actually relying on the bootloader to take over and handle loss of a watchdog reset, and the only thing that knows to do... is typically reset. Thats outside the marlin code.